florentbr/OWON-VDS1022 #1: ARM\RPi compatible

apocalypse2012, 2020-01-02:
This project is just what I am looking for. I am building a small hobby electronics home-lab, and using an Owon VDS 1022i as my scope. Only thing in my way is Arm compatibility. I am using an RPi as my platform here for both project and hopefully to host this scope. Any possibility of getting the linux setup to install on a Raspberry Pi?

Cheers,
Raymond

CommonGrounds, 2020-01-02:
Install procedure for Raspberry Pi 4.

[https://www.eevblog.com/forum/testgear/owon-vds1022i-quick-teardown-(versus-the-hantek-6022be)/528/](url)

florentbr, 2020-01-03:
I added the armhw/arm64 native libraries and updated the script (release 1.0.30-cf7). It should now work with your hardware.

apocalypse2012, 2020-01-03:
That is awesome news. Thank you. I will respond again when I get it working on my unit.

Cheers

foice, 2020-01-26:
I do not have the OWON oscilloscope (yet) but I have tested it compiles as described here (after a bunch of JRE packages have been installed) on my RaspberryPi2 running `Raspbian 8.0 jessie`
`Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux`

Of course it shows nothing on the screen as `no available USB found`. Questions: does this guarantees that this software will work with the OWON hardware? I am puzzled by the menu "Install USB driver", is it really a thing to install ? (and check it installs succesfully before buying)

florentbr, 2020-01-26:
@foice, the software works on all platforms as long as Java 8 or superior is available along with the native library for libusb.

Once installed, launch the app from a terminal by executing `owon-vds-tiny`. If there's no error, then the app works with your hardware.

Regarding the menu "Install USB driver", it is not needed on Linux.

apocalypse2012, 2020-01-26:
I was remiss in not coming back earlier. Regarding the armhw/arm64 native libraries and updated the script (release 1.0.30-cf7). It all worked perfectly on my Raspberry Pi 3.
So now my RPi is a my dedicated bench computer, running the interface for my oscilloscope and serving my Octopi server for my 3d printer. All that in addition to occasional duty as a project board from gpio control of circuits. This library is a big win in helping me base a lot of my hobby work on the RPi.

closed 2020-01-26 by florentbr