florentbr/OWON-VDS1022 #2: Can not connect to device

meibye, 2020-01-11:
Thanks for making this project.

I am working on installing "release 1.0.30-cf7" on Windows using the install-win.cmd. It have been necessary to make a small update to your script - adding the quotes around '%~dp0.'. The updated command below
FOR %%I IN ("%~dp0.") DO SET THIS_DIR=%%~fI

This change is needed as the path for file names may contain spaces.

I can get the OWON application up and running but the application can not find the USB device. However, using the official application it works. Do you have any idea of what to look for?

florentbr, 2020-01-11:
I noticed that the device doesn't connect on Windows 64bits when the installed Java is 32bits.
Try this modified [install-linux.sh](https://textuploader.com/165tg) and let me know if it fixes the issue.

meibye, 2020-01-12:
Thanks Florent,

 

That did the job. It works.

 

I have only 2 error messages during install. These are related to the command given below. I do not know if it is required to use the dll’s which are part of the package.

 

D:\Michael\Owon oscilloskop\OWON-VDS1022-1.0.30-cf7>XCOPY /D /Y /C ".\lib\win\x86\msv*100.dll" "C:\WINDOWS\System32\" 1>NUL

Sharing violation

Sharing violation

florentbr, 2020-01-12:
> I have only 2 error messages during install. These are related to the command given below. I do not know if it is required to use the dll’s which are part of the package.
>
> D:\Michael\Owon [...]

You can ignore this error. It simply tries to add the Microsoft C Runtime Library in case it's missing. In your case, you already have it, but it is currently used by another process, thus it can't be updated with a more recent version.

florentbr, 2020-01-13:
fixed in release 1.0.30-cf8

closed 2020-01-13 by florentbr