florentbr/OWON-VDS1022 #3: Can not select options in  combo boxes

meibye, 2020-01-12:
When I select the small triangle pointing downwards in a combo box such as Mode in the Trigger function, I get the foillowing options presented Edge, Video, Slope, Puls. However, when I try to move the cursoe down the combo box either just closes or it modifies the presented options to only Edge, Video. Hence, it is not possible to select e.g. Slope. The same holds for other combo boxes, e.g. Language in Utility, it starts by showing all languages then reduced the selection set to the top 2.

It does not seem to be the case with all combo boxes, but most of them.

florentbr, 2020-01-12:
I can't reproduce the issue on Windows 10 or Linux.  
Which version of Windows are you using  ?  
Which version of Java have you installed (`%JAVA_HOME%\bin\java.exe -version`) ?  
Can you provide a screenshot ?

meibye, 2020-01-12:
I am using Windows 10 Pro version 1909

The Java version is given by the information below.

java version "1.8.0_231"
Java(TM) SE Runtime Environment (build 1.8.0_231-b11)
Java HotSpot(TM) Client VM (build 25.231-b11, mixed mode)

Finally I have attached a small MP4 video in inside the zip showing the problem. I hope you can see it.
[OWON VDS1022 combo error_1.zip](https://github.com/florentbr/OWON-VDS1022/files/4050736/OWON.VDS1022.combo.error_1.zip)

florentbr, 2020-01-13:
Fixed in release 1.0.30-cf8

closed 2020-01-13 by florentbr

meibye, 2020-01-14:
It works. Thanks