florentbr/OWON-VDS1022 #6: Java Runtime not found

Pepe4All, 2020-03-29:
On the console this shows up even though I updated it and then reinstalled it:
Check driver ...
Get Java from JAVA_HOME ...
Locate Java in PATH ...
Error: Java Runtime not found.
  Environement variable "JAVA_HOME" is not set or invalid and "PATH" doesn't point to `java.exe` .
  To install, visit  adoptopenjdk.net  or  java.com

florentbr, 2020-03-30:
What's your OS and which installer did you run to install Java ? 

Looks like your Java installer didn't update the PATH environment variable. If you installed it with adoptopenjdk.net, then make sure to select "Add the installation to the PATH environment variable".

You could easily fix it manually. Google "Setting the JAVA_HOME Variable in Windows".

Pepe4All, 2020-04-10:
I fixed the Java Runtime not found, but have a different problem:

I opened the .bat file and ran it as an admin, then I got this on the screen:
Check driver ...
Get Java from JAVA_HOME ...
Switch to 32 bits install ...
Check driver ...
Get Java from JAVA_HOME ...
Install Microsoft C Runtime Library dependency ...
Install program files ...
Clear settings ...
Create uninstall and menu shortcut ...
Done !
Press any key to continue . . .

Then I checked %appdata% file and there wasn't OwonVdsTiny file and there weren't any changes in the software either.
What can I do to fix this?

Thanks in advance!

P.S Sorry for the late reply, I was out of town and couldn't manage to reply earlier.

florentbr, 2020-04-10:
From the log, the app was successfully installed. You may have to restart your machine to see the shortcut in the Windows menu.

> Then I checked %appdata% file and there wasn't OwonVdsTiny file and there weren't any changes in the software either.
> What can I do to fix this?

What issue are you trying to fix ? Files in `%APPDATA%\OwonVdsTiny` are created by the application itself once it is launched.

Pepe4All, 2020-04-10:
There isn't a OwonVdsTiny in the %appdata% folder:

https://user-images.githubusercontent.com/62845080/78991094-3463b480-7b49-11ea-9524-460ac463d6f8.png

I ran it a couple of times and tried restarting my computer, but there aren't any changes in the software.

florentbr, 2020-04-10:
In your case files are installed in `C:\Program Files (x86)\OwonVdsTiny` not in `%APPDATA%\OwonVdsTiny`.
Please provide a clear description of your issue with detailed steps and expected outcome.

Pepe4All, 2020-04-10:
I found it, everything works now.
Thanks for your help!

closed 2020-04-10 by florentbr