florentbr/OWON-VDS1022 #7: Windows version does not work with version greater than OpenJDK 8 (LTS)

csdpm, 2020-06-06:
Firstly, thanks for a great quality enhancement of the VDS1022 software!

There seems to be a compatibility issue with the OpenJDK version...
In the Dependencies section of your README.md is states: "This software requires Java Runtime Environment 8 or superior".  The https://adoptopenjdk.net page currently defaults to OpenJDK 11 (LTS).  I have found that with this version, your software (although it installs completely) fails to recognise the VDS1022 Hardware, even though it appears correctly in the Device manager.

Uninstalling OpenJDK 11 (LTS) and manually selecting and installing OpenJDK 8 (LTS) resolves the problem.

Tested in Win 10 environment only.

florentbr, 2020-06-06:
Looks like the flag `java -d64` is no longer present with JAVA OpenJDK 11 on Windows. It is used by the installer to determine the bitness of Java. Without this flag, the script installs the 32bit libraries which can't be used by Java 64bit.

To overcome this issue, just comment the code from [line 44  to 48](https://github.com/florentbr/OWON-VDS1022/blob/1.0.30-cf9/install-win.cmd#L44).

csdpm, 2020-06-06:
Many thanks for the very fast explanation and resolution Florent. :)