florentbr/OWON-VDS1022 #9: OWON-VDS1022 Language Rus

LeonIndman, 2020-06-14:
Hi florentbr!
Thank you for your great work on improving the software of this excellent oscilloscope!
I would also like to add my small contribution to the improvement of this software. I suggest
you add to your distribution a corrected translation into Russian, as well as translation into Russian of tips. I published an archive with these files in this thread
https://www.eevblog.com/forum/testgear/owon-vds1022i-quick-teardown-(versus-the-hantek-6022be)/msg3096053/#msg3096053
I also wanted to draw your attention to a few points in the shell that I would like to fix. If you have the opportunity and desire to do this?
I took screenshots of these flaws here
https://www.eevblog.com/forum/testgear/owon-vds1022i-quick-teardown-(versus-the-hantek-6022be)/msg3093578/#msg3093578
Thanks!

florentbr, 2020-06-14:
Hi and thanks for contributing to this software. I'll add the translation
in the next release.
Could you send me the related files you've modified?
Regarding the Mark Cursor, I'd argue that the drop-down list is correctly
placed. The horizontal Cursor can only display voltages from a single
channel at time. The vertical Cursor on the other hand displays time
information regardless of the selected channel.

Regards,
Florent

florentbr, 2020-06-14:
Hi and thanks for contributing to this software. I'll add the translation
in the next release.
Could you send me or attach the related files you've modified?
Regarding the Mark Cursor, I'd argue that the drop-down list is correctly
placed. The horizontal Cursor can only display voltages from a single
channel at time. The vertical Cursor on the other hand displays time
information regardless of the selected channel.

LeonIndman, 2020-06-14:
Thanks for the answer! You are right about the mark cursor.
I attach Russian translation and help files.

> Hi and thanks for contributing to this software. I'll add the translation
> in the next release.
> Could you send me or attach the related files you've modified?
> Regarding the Mark Cursor, I'd argue [...]
 
 
Best regards,Leon

florentbr, 2020-06-30:
Still no file?

LeonIndman, 2020-06-30:
[RussianVDS1022.zip](https://github.com/florentbr/OWON-VDS1022/files/4853909/RussianVDS1022.zip)

LeonIndman, 2020-06-30:
Sorry, I didn’t immediately see that the language file is missing

florentbr, 2020-06-30:
I noticed that your MsgLib_ru.properties is not UTF8 encoded and some translations are missing compared to the latest release. Could you update the latest [MsgLib_ru.properties](https://github.com/florentbr/OWON-VDS1022/files/4854022/MsgLib_ru.zip) ?
Thanks

LeonIndman, 2020-07-01:
I updated the latest file MsgLib_ru.properties
Some measurement parameters, for example Vpp, Vmax, Vmin and similar, are best left in the English version, since the Russian translation only distorts these concepts.
[MsgLib_ru.zip](https://github.com/florentbr/OWON-VDS1022/files/4858796/MsgLib_ru.zip)

florentbr, 2020-12-13:
added to release 1.0.33-cf11

closed 2020-12-13 by florentbr