florentbr/OWON-VDS1022 #10: Question: would this work with VDS6102 ?

flytrex-vadim, 2020-06-30:


florentbr, 2020-06-30:
It won't work with the VDS6102. The software is different.

closed 2020-06-30 by florentbr

paulojarasantibanez, 2020-12-23:
I'm using your vesión driver/software from 1.0.30-cf10 till the new update and works perfect! on my Linux kubuntu 18.04.5 with kernel  4.18.0-17. you really save my life to avoid move to windows.
so, really thanks for that.
i thinking to buy the new VDS6102 with wifi. to avoid grounding issue, take 100Mhz and 1G sample advantages.  i know that the current version doesn't support this new device. so, right to the point could you add this compatibility to current version or make a new development? can i help you $$$ i am not programmer so my only way is some $$$

florentbr, 2020-12-26:
@paulojarasantibanez, porting the VDS6102 software to Linux is possible since the required NI-VISA driver is available on Debian. I can build a modified version of the VDS6102, but the number of tweaks I can add from the VDS1022 depends on how much you are willing to spend. I don't own this device, so I'm not interested in spending free time on the software.