florentbr/OWON-VDS1022 #11: Compatibility with SDS and LAN

priitv8, 2020-09-11:
Hi,

I own a SDS5032E oscilloscope. 
https://owonna.com/products_owon_sds-e_series_digital_oscilloscope
Like with so many of users, my problem is that the original software only works on windows and I am a Mac user.
Hence, I have been looking for a Mac-solution and indeed have found these Java ports to Mac.
Now, I have 2 questions:
1) what would it take to make this client also compatible with SDS5032E?
2) why is the LAN connection disabled in all these client ports? As far as I can tell, the LAN protocol works like USB one, and is actually also available here, on GitHub: https://github.com/bbum/ow

Thank you
Priit

florentbr, 2020-09-12:
Hi, the software for the VDS series is not the same as the one for the SDS
series, but it looks like it shares some of the code. You'll have to
decompile the binaries with JD Decompiler, modify the code and recompile.
It's not that hard but it's time consuming.

Regarding the Lan connection, I only own a VDS1022 with no Lan connector.
So I can't tell why it's disabled.

Regards,
Florent

florentbr, 2020-09-12:
Hi, the software for the VDS series is not the same as the one for the SDS series, but it looks like it shares some of the code. You'll have to decompile the binaries with JD Decompiler, modify the code and recompile. It's not that hard but it's time consuming.

Regarding the Lan connection, I only own a VDS1022 with no Lan connector. So I can't tell why it's disabled.

closed 2020-09-12 by florentbr

priitv8, 2020-09-13:
So what you say, is that you have to decompile Java binaries? 
I read that some early version had source JAR bundled in, do you have access to that?
If I were to modify the code myself, can you hint me, which Class  handles the USB detection device detection?
I tried that 1022 software on windows, it correctly sees my SDS on USB bus and attempts to connect, but then hangs there.

Menu seems to be disabled exactly because 1022 does not have a LAN interface. But it is noteworthy, that it is in there.
Software supplied for my SDS is actually built on same eclipse framework but I have yet to make it launch under macOS.
At the moment I get exception:
java.lang.NoClassDefFoundError: org/eclipse/swt/widgets/Display

Trying to look into it.
I would still like to get your codebase running, because I like its UI so much better, than mine:
http://files.owon.com.cn/software/pc/OWON_Oscilloscope_Setup.zip

So help with possible source code and pointers to which classes to start from, would be very welcome.
Thank you for your support.

florentbr, 2020-09-13:
I've made too many changes across all the classes. My sources won't be useful since the SDS software is different.

Start by extracting your installer.
Replace the ./jre with a release of JRE 6 for MacOS
Add the ./lib folder from this project
Move all the ./plugins/**/*.jar into ./plugins/ on a single level

Then try to launch the app :

	jre/bin/java -Djava.library.path="lib/mac/x86_64" -cp "plugins/*" com.owon.uppersoft.oscilloscope.Main

If there's an exception then decompile the class raising the exception and fix it. With the VDS serie I had to fix these classes :

		--- a/com/owon/uppersoft/vds/core/usb/CDevice.java
		+++ b/com/owon/uppersoft/vds/core/usb/CDevice.java
		@@ -604,11 +604,6 @@ public class CDevice {
		 	 */
		 	private void claim_interface(long usb_dev_handle, int configuration,
		 			int interface_, int altinterface) throws USBException {
		-		if (LibusbJava.usb_set_configuration(usb_dev_handle, configuration) < 0) {
		-			usbDevHandle = 0;
		-			throw new USBException("LibusbJava.usb_set_configuration: "
		-					+ LibusbJava.usb_strerror());
		-		}

		--- a/com/owon/uppersoft/vds/util/SystemPropertiesUtil.java
		+++ b/com/owon/uppersoft/vds/util/SystemPropertiesUtil.java
		@@ -52,15 +52,12 @@ public class SystemPropertiesUtil {
		 	public static final Font FONT_LIST[] = GraphicsEnvironment
		 			.getLocalGraphicsEnvironment().getAllFonts();
		 
		-	public static final int ARCH_NUM = Integer.parseInt(OS_ARCH.substring(1));
		-	public static final String WIN_NAME = OS_NAME.substring(8);
		-	public static final double WIN_VER = Double.parseDouble(OS_VERSION);
		 
		 	public static void main(String[] args) {
		 		SystemInfo();
		 
		-		System.out.println(ARCH_NUM);
		-		System.out.println(WIN_NAME);
		-		System.out.println(WIN_VER);
		+		System.out.println(OS_NAME);
		+		System.out.println(OS_ARCH);
		+		System.out.println(OS_VERSION);
		 	}
		 }

Recompile only the modified classes and replace them manually in the original .jar. You'll have to use a Java 6 SDK since it will fail with the later versions:

	javac -cp "plugins\*" myclass.java

priitv8, 2020-09-14:
Hi.

Thank you, these tips are really helpful!
Now I managed to launch the GUI, using Java 8 and org.eclipse.swt.cocoa.macosx.x86_64-3.105.3.jar
I had to add one additional parameter to VM: -XstartOnFirstThread

I did exactly according to your advice and copied lib/mac/x86_64 to jre folder, but it still fails to find the usbJva in the library path.

Exception in thread "Thread-1" java.lang.UnsatisfiedLinkError: no usbJava in java.library.path

What am I missing here? (incidentally, my plugins/ch.ntb.usb_0.5.9 folder contains exactly the same jnilib and dylib files as the lib/mac/x86_64)

[image]

florentbr, 2020-09-14:
move the `lib` folder to the current folder next to `plugins`.

priitv8, 2020-09-14:
Thanks! That worked!
Now I need to dig into the code, as the problems with communicating with device seem to be buried there.
I can work on downloaded waveforms in .bin files, though.

[image]

priitv8, 2020-09-14:
First exception seems to be in the same SystemPropertiesUtil.class that you listed above:
Exception in thread "Thread-5" java.lang.ExceptionInInitializerError
	at com.owon.uppersoft.oscilloscope.communication.lan.RapidLANCommunication.connect(RapidLANCommunication.java:62)
	at com.owon.uppersoft.oscilloscope.communication.lan.RapidLANCommunication.startInit(RapidLANCommunication.java:41)
	at com.owon.uppersoft.oscilloscope.communication.BetaUploadFrame.progressUpload(BetaUploadFrame.java:163)
	at com.owon.uppersoft.oscilloscope.communication.BetaUploadFrame.run(BetaUploadFrame.java:138)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NumberFormatException: For input string: "86_64"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Integer.parseInt(Integer.java:580)
	at java.lang.Integer.parseInt(Integer.java:615)
	at com.owon.uppersoft.common.utils.SystemPropertiesUtil.<clinit>(SystemPropertiesUtil.java:60)

Where line 60 tries to assign "86_64" to an integer:
60: public static final int ARCH_NUM = Integer.parseInt(OS_ARCH.substring(1));

priitv8, 2020-09-17:
Can you imagine, fixing just this one class SystemPropertiesUtil.java as you suggested, brought my app to running condition.
USB obviously does not work. But over LAN connection I am able to download binary data, deep memory and screen bitmaps.
I am also able to use the device control panel and SCPI commands!
[image]
[image]

florentbr, 2020-09-17:
I'm glad it worked for you. USB should work once you have fixed your CDevice.class .

priitv8, 2020-09-30:
I found something kind-of-a-SDK for VDS6000 :)

http://files.owon.com.cn/software/Application/VDS6000_Secondary_development_documents.zip

Source: https://owonna.com/products_owon_vds6000_series_pc_oscilloscope