florentbr/OWON-VDS1022 #12: Linux install issue in cf10 - unsure if it is me or the program

ExordiumAndTerminus, 2020-10-15:
I am having trouble running this program on my Elementary OS Linux build.  I ran the linux install sh script setup through sudo apt install and it seemed to work fine.  However, when I tried to execute it from the icon, it would act like it was doing something then disappear from the task bar.  

Where did I go wrong from the install? It installed Java 11 as well.  I did not plug in my owon scope yet, but figured the program should still function while disconnected...

Steps tried to remedy issue:

- Uninstalling and reinstalling a few different times
- Executing 'owon-vds-tiny' through the terminal while in the program's main folder
      ** this gave me the the error: 
                    "Error: Could not find or load main class com.owon.vds.tiny.Main
                     Caused by: java.lang.ClassNotFoundException: com.owon.vds.tiny.Main".  
- 'jar xvf'ing all the packages under the jar folder, then typing "java 'com.owon.vds.tiny.Main'"
     ** allowed me to actually pull up the main program, but I doubt functionality is all there, as there are still usb loading errors in the terminal, repeatedly showing up - most notably this one: java.lang.NoClassDefFoundError: Could not initialize class ch.ntb.usb.LibusbJava

Thanks in advance for the help!

florentbr, 2020-10-15:
I think the issue is related to file permission. 
Could you try this change in `install-linux.sh` line 327 and 330 :
```
324   copy () {
325       if [[ "$1" == */ ]] || [[ -d "$1" ]] ; then
326           mkdir -p "$1"
327 -         cp -r "${@:2}" "$1"
    +         cp --no-preserve=mode,ownership -r "${@:2}" "$1"
328       else
329           mkdir -p "$(dirname "$1")"
330 -         cp "${@:2}" "$1"
    +         cp --no-preserve=mode,ownership "$2" "$1"
331       fi
332   }
```

The modified `install-linux.sh` :
https://raw.githubusercontent.com/florentbr/OWON-VDS1022/32357e8ffd350a6ac6d5b57d9eb0b73209f2e1d7/install-linux.sh

ExordiumAndTerminus, 2020-10-15:
Hey that worked!  Thanks for the help!

florentbr, 2020-12-13:
fixed in release 1.0.33-cf11

closed 2020-12-13 by florentbr