florentbr/OWON-VDS1022 #14: Write calibration data

nico202, 2020-11-13:
Hi!  For some reason my VDS1022I requires an auto-calibration every time I run it.

I found that the problem might lie in the stored flash data (the program prints -86, 85, and then 2000 time -1).

Can you anyone send my what's yours printing? I was able to manually send (via libusb) the required message to write data, but I don't know which values should be in this array.
If anyone could send me the log, I'd be really grateful!

Thanks, Nicolò

florentbr, 2020-11-13:
Here's a dump of the flash from my VD1222I:
[flash-vds1022i.zip](https://github.com/florentbr/OWON-VDS1022/files/5539574/flash-vds1022i.zip)

    Offset  Type        Name              Value (little endian)
    ------  ----------  ----------------  ---------------------------------------------------------------------------
    0       uint16      Flash header      0x55AA  (-86 85)
    2       uint32      Flash version     2

    *** Factory Calibration ***
    6       uint16[10]  CH1 Gain          0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    26      uint16[10]  CH2 Gain          0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    46      uint16[10]  CH1 Amplitude     0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    66      uint16[10]  CH2 Amplitude     0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    86      uint16[10]  CH1 Compensation  0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    106     uint16[10]  CH2 Compensation  0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v

    *** Registry ***
    206     byte        OEM               0 or 1
            char*       Device version    null terminated string - ASCII encoded
            char*       Device serial     null terminated string - ASCII encoded
            byte[100]   Locals            0 or 1 for zh_CN, zh_TW, en, fr, es, ru, de, pl, pt_BR, it, ja, ko_KR, ...
            uint16      Phase fine

    *** Device Calibration ***
    1006    uint16[10]  CH1 Gain          0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    1026    uint16[10]  CH2 Gain          0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    1046    uint16[10]  CH1 Amplitude     0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    1066    uint16[10]  CH2 Amplitude     0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    1086    uint16[10]  CH1 Compensation  0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v
    1106    uint16[10]  CH2 Compensation  0-2048  for  5mV, 10mV, 20mV, 50mV, 100mv, 200mv, 500mv, 1v, 2v, 5v

closed 2020-11-13 by florentbr

nico202, 2020-11-13:
Wonderful, I'll try it tomorrow, but is exactly what I needed, thanks!

nico202, 2020-11-14:
Yeah that worked :) Thanks also for the data structure