florentbr/OWON-VDS1022 #19: German documentation for Owon VDS1022(I)/VDS2052(I)

km-linux, 2021-02-03:
I'm writing an Inoffical German Manual for the Owon VDS1022(I)/VDS2052(I).
The Manual is nearly ready.

First I made is an English LibreOffice Writer Document, which I copied and translated into German. From these documents it's easy to make PDFs or translations into other languages.

Are you interested?

Regards from Germany

florentbr, 2021-02-03:
Hi, I can add your manual if you think it can help new users.
The manual is accessible via `F1` if Deutsch is selected and if `doc\manual_de.pdf` is present. If not present, it opens `doc\manual_en.pdf`.

km-linux, 2021-02-03:
Hi, is it the same with ../doc/tips_en and ../doc/tips_de?
I am currently modifying the images. Must anything be changed in the htm-files?
When it is ready I will load it up.

florentbr, 2021-02-03:
Yes it's the same with the tips. It first looks for `./doc/tips_de` and falls back to `./doc/tips_en` if not present. I would just translate the shortcuts, the images are pretty much useless. I'll probably remove them in the future to only keep the shortcuts.

km-linux, 2021-02-03:
I changed the first 3 images and copied it to ./doc/tips_de and it is OK. 

I don't think, that the images are pretty much useless. I bought my OWON three years ago and did not use it. From the past I know only the analog oscilloscopes. 
The handling of a PC oscilloscope is different, as there are no buttons on the front. If someone is new with this oscilloscope, it's helpful to have got the pictures. Leave the images in the software.

florentbr, 2021-02-03:
I meant to say that the images in the tips are useless since they are already present in the manual via F1. All the oscilloscopes works pretty much the same way. If a new user can't find his/her way around with a simple interface as this one, they will eventually end-up reading the manual or a tutorial.

km-linux, 2021-02-05:
I developed software, wrote books and was teaching in the past. My experience is that you can't change people.
Leave the pictures in, they are helpful for beginners and they don't bother anyone.

This help is now ready for german. I have uploaded it to you as a ZIP file. For English I have also attached a ZIP_file. 
There were errors in the htm-files everywhere. The Chinese have entered charset=gb2312. This is a character set for simplified Chinese characters. I translated with Google the Chinese text in the title: "Untitled document". Someone didn't know exactly what they were doing.

tips_de and tips_en now have the utf8 character set and the Chinese characters in the title are deleted. This is necessary to display special german characters (ÄäÖöÜüß). utf8 has also characters for other languages. Please import also the new tips_en, otherwise there are always problems with translations. With a text editor you can edit the files with errors (charset=gb2312), but with geany it s not possible to save files with charset=gb2312 and special characters.

tips_ru I did not change, because I can not check the effects.

In the moment I'm searching how to upload the files. I I find it, I will do so.

Regards

km-linux, 2021-02-05:
Here are the links to download the ZIP-files:

https://mun-steiner.de/download/owon/OWON_VDS_tips_de.zip
https://mun-steiner.de/download/owon/OWON_VDS_tips_en.zip

florentbr, 2021-02-05:
From my experience, users don't pay attention to information they are not looking for and don't like to be enforced with them. Microsoft has been trying for years to push tips/suggestions though Clippy/Cartona and failed. But what bothers me the most is that these images and text contents become blurry when Windows is scaled over 100% on an FHD screen. It makes a bad user experience. Anyway, I'm not planning to touch it any time soon. I'd rather spend some time on virtually increasing the number of samples or on enabling triggers with higher time bases.

This software is a relic from the past when IE6 was the main cursing topic among web developers. The original code is ANSI encoded. I've already converted the internal translation from ANSI to UTF8. It's zipped in 
owon-vds-tiny-1.0.33-cf11.jar in src\com.owon.vds.tiny\com\owon\uppersoft\dso\i18n\MsgLib_de.properties if you want to have a look. FYI, it's possible to directly upload a zip in a comment here on github.

km-linux, 2021-02-05:
I have never done anything with java, so looking at the code is of no use.
Another idea: Swap the tabs so that "Shortcuts" is the first tab. Then these are always displayed first.
The German manual is not quite finished yet. I will contact you when I am ready.

florentbr, 2021-02-05:
The translations are stored in text files, not Java code. A `.jar` file is just a zip file. These are the files in question in case you want to fix some translations: [i18n.zip](https://github.com/florentbr/OWON-VDS1022/files/5933642/i18n.zip)
I agree, it would be better to have the shortcuts first.

km-linux, 2021-02-05:
I had a look into the i18n.zip file. I will do some corrections and you'll get the file in the next days.
It's important, that the charset is everywhere utf8.

Please check the tips_ru, where I did not change anything. I think you can change charset to utf8 and remove the Chinese signs in the title, like I did it in tips_en and tips_de. 

shortcuts.htm: I searched with Google "russian charset utf 8"
I found: https://www.utf8-chartable.de/unicode-utf8-table.pl?start=1024

km-linux, 2021-02-05:
in MsgLib.properies is a block starting:

FROM
_**Internal.**_ApplyToMachine           = Apply To Machine
TO
Internal.savetxt                  = Export

Is this only for **internal** use? 
Will any user see the messages?

In the German file there are a lot of Chinese signs. If nobody will see it, I can copy the English text.

florentbr, 2021-02-05:
It's easy to switch from one encoding to another with notepad++. It's just two clicks away via the dedicated menu.

Regarding the translation, some of the internal labels are used, some are not. I translated some of them for the calibration dialogue which was originally hidden. I would only bother with the labels that are already translated in English in MsgLib_en.properties (same as MsgLib.properties). So the labels with Chinese characters in MsgLib_de.properties are most likely not used, unless I forgot to translate some of them.

km-linux, 2021-02-07:
Today I found something interesting. The Owon VDS1022(I)/VDS2052(I) is sold as an other brand, as OWON doesn't sale the old one. 
Here it is:
https://www.reichelt.de/de/en/pc-oscilloscope-25-mhz-2-channel-with-usb-peaktech-1290-p141197.html?&trstct=pol_7&nbc=1
https://www.peaktech.de/productdetail/kategorie/pc-oszilloskope/produkt/p_1290.html
Description: **These newly developed** 2-channel PC oscilloscopes ......
Everything is with the old OWON software, with Chinese charset.  :-))

Now to my development:
1.  I am checking my translated manual against the German manual of Peaktech. 
2.  Next I will test everything with my device.
3. Then I make the corrections in the language files and test it.
4. Last is to correct my manual again, as there are new images with new text. 

Regards

km-linux, 2021-02-08:
I changed different labels and descriptions in the File "MsgLib_de.properties" and did the new one into the file "owon-vds-tiny-1.0.33-cf11.jar".  Then I tested the display panel.

The displayed fonts in the Home menu are very big and bold. The result is, that the words sometimes are not displayed completely and cut off at the right side. The problem is, German words are longer.

**Is it possible to make the font smaller or not so bold?**

Regards

florentbr, 2021-02-09:
> Is it possible to make the font smaller or not so bold?
I'd rather increase the width of the Home Page. How long is missing?

km-linux, 2021-02-09:
I must try, when it's changed. I think, add 15% width.
Is it inside the java code?
When you changed it, send the code to me, that I can test it, using the modified "MsgLib_de.properties".

florentbr, 2021-02-10:
Yes it's inside the Java code. I'll adjust the width if it's not too long. It'd be better if you could find something shorter. Even the Russian translation which is usually longer fits easily in the home page.

km-linux, 2021-02-10:
I have already shortened and will also partially leave English terms in. I can describe these in the manual.
Nevertheless, it is currently still a bit too narrow, even in English:

Panel **Math** select **FFT**:
Frequency Base: Behind the **pull down menu** I can read " / " only
Inside the pull down there menu is a space in front of **HZ** . 
**I think there should be no space** and if you like to make a space, it must be behind the numbers. With no space it's smaller.

Panel **Display**
The buttons **Vector / Dots** are too small. 
The size must be the same like the buttons in the Trigger panel.

Panel **Trigger**
The right **Hold Off** pull down menu is too small. 
If **+++** is selected, **+..** is displayed.

When the panel is 15% wider some other problems are fixed.
See Panel **Measure** there is **+PulseWidth** cut off to **+PulseWi...**

km-linux, 2021-02-11:
Here are the screenshots concerning my last mail:
[image]
[image]
[image]

km-linux, 2021-02-16:
Actually I'm writing at the German manual. The old OWON manual is worse, as parts like probe compensation are missing and there is no structure inside. I want to have a structure based on your menus and I also want to replace the old images that are no longer correct.

When this is all done, I want to implement it in the English manual. Then we will have two new manuals that are also up to date with the current software. You will get the manual as draft to make corrections and later on you will have got an actual English manual, too.

**If you have time it would be nice if you could make the changes in the Java code.** I have already described the details. This would save me a lot of work, because I could immediately add the current images to the documentation. 
If the panel is 15% wider, it's fine. 20% is better, but the decision is yours.

In the Owon manual I found on page 28 **Save/Refer**. I can not find it in your software. Did it get an other name or does it not exist anymore?

Regards

florentbr, 2021-02-16:
I made significant changes to the interface which is not yet finalised. I'll let you know when it's done. 
It would be great if you could separate the capture from the arrows/texts to make it easier to replace it in case of major changes. Regarding the "Save/Refer" from the Utility menu, I moved it to its own menu "Save / Pin" in the Home page.

km-linux, 2021-02-16:
I don't understand what they mean:
**It would be great if you could separate the capture from the arrows/texts.**
Do you mean the documentation, to separate images and text in the images?

florentbr, 2021-02-16:
I'm talking about the arrows and texts in the documentation which are on top of the screenshots.

km-linux, 2021-02-16:
Do you mean this?

[image]

I write the German manual with LibreOffice Writer and use tables. So I can separate images and text for different translations.

km-linux, 2021-02-17:
**Here are the tips files.** 
I delete the message containing the old download links above.

English tips: 
[OWON_VDS-tips_en.zip](https://github.com/florentbr/OWON-VDS1022/files/5994469/OWON_VDS-tips_en.zip)

German tips:
[OWON_VDS_tips_de.zip](https://github.com/florentbr/OWON-VDS1022/files/5994473/OWON_VDS_tips_de.zip)

**Special German tips files for translation.** 
**There are files for GIMP (xcf) where images and text are separated.** After changing them in another language, you can simply export them as a png file. Then the images are quickly and easily ready. 
[OWON_VDS-tips_de_translate.zip](https://github.com/florentbr/OWON-VDS1022/files/5994522/OWON_VDS-tips_de_translate.zip)

km-linux, 2021-02-20:
You wrote:
**It would be great if you could separate the capture from the arrows/texts.**

I tried different  results:
The arrows are UTF8 characters. One is done in LibreOffice and the other is the result in PDF.
**The images and text are consistently divided into tables.**
[Example_Table_PDF.zip](https://github.com/florentbr/OWON-VDS1022/files/6016139/Example_Table_PDF.zip)

What do you think about it?

km-linux, 2021-02-23:
I am checking everything vs the old OWON Manual. 
**I found a difference.**

[image]
[image]

OWON has **vertical rose lines** in their image, when switching on **Frequency**.
**Horizontal lines** are displayed when switching on **Amplitude**.
[image]

**These lines are not displayed in your software.
I this correct?**

km-linux, 2021-02-25:
Hi Florent,
the new English / German manuals are 90 % ready. They are completely new, images and text are completely separated. 

I have a few more questions about the meaning of buttons: 

**Utility Menu -- Settings:**
**Default:**     Default values of OWON ??
**Save:**     Save the values of auto-Calibrate ??
**Restore:**     Restore the saved values of Auto-Calibrate ??
**Auto-Calibrate:**     Runs Auto-Calibrate ??

km-linux, 2021-02-27:
Here are 2 zip files for the language English and German to patch **owon-vds-tiny-1.0.33-cf11.jar**
The name of the zip files are, the place to patch the java file.
[com_owon_uppersoft_dso_i18n.zip](https://github.com/florentbr/OWON-VDS1022/files/6054133/com_owon_uppersoft_dso_i18n.zip)
[com_owon_vds_i18n.zip](https://github.com/florentbr/OWON-VDS1022/files/6054134/com_owon_vds_i18n.zip)

- **MsgLib_en.properties:**  Only some lines are changed from Chinese to English
- **MsgLib_de.properties:** German now correct
- **messages_de.properties:** German NEW

florentbr, 2021-03-02:
I've released the modified interface and included your translation. I'll add the documentation in the next one once it's finalised.

> I tried different results:
> The arrows are UTF8 characters. One is done in LibreOffice and the other is the result in PDF.
> The images and text are consistently divided into tables.
> Example_Table_PDF.zip
> 
> What do you think about it?

I'd rather have the long arrows from the original doc with a short table describing the numbers.
The short arrows you've used makes it hard to see what it's pointing to and the long table with images forces the reader to scroll back and forth. In LibreOffice, unwrap the image (Format>Wrap>None), insert the arrows and text areas on top (Insert>Shape>Line or Text Box) and group them so that they can be moved altogether.

> But I have a few more questions about the meaning of buttons:
> 
> Utility Menu -- Settings:
> ... 

The "Default" button restores the factory settings and calibration or loads it from a file depending on the choice in the presented dialogue.

The "Restore" button loads the default settings (voltbase, timebase...) but keeps current the calibration. It's the same as the "Restore" button at the bottom/right of the chart. It jumps back to the Home Menu because it reloads the dock once done.
The "Save" button overwrites the default settings with the current settings. It's meant to personalise the settings.

km-linux, 2021-03-04:
Looked at the updates in your software and found that I need to change some in the manual. 
With the arrows that I still will change. I didn't like it too, but it was the first draft to separate images and font.

There is something I do not understand. 
In the "**Channel**" menu there is now a "**Current**" field. 
**What does that do? 
Why did you change it?**

There were some issues here where you added some functions and explained them. 
**If it is necessary, we could add another chapter before "Technical Specifications" where different things are explained.** 
These texts would have to come from you, because I can't explain it. I will do the translation into German afterwards.

When I translated the manual I was looking for a very good translator who could handle the technical texts. 
I can recommend **www.DeepL.com/Translator (free version) https://www.deepl.com/translator**. 
It is very good and saves you a lot of typing, because you can do a lot with copy & paste. 
More than 96% is translated correctly. I also use it for this as it is just faster.

km-linux, 2021-03-04:
I will change the documentation.

In "**Technical Specifications**" is written: **Communication port  USB1.1**
**That is wrong.**

I checked it by command **lsusb -v**
**Communication port is USB2.0**

florentbr, 2021-03-04:
> Looked at the updates in your software and found that I need to change some in the manual.

That's what I meant when I said I made significant changes to the interface.
Sorry that made you work on obsolete screenshots.

> With the arrows that I still will change. I didn't like it too, but it was the first draft to separate images and font.

It'd be great if you could make it light and simple like the original one, something like [that](https://docs.google.com/document/d/1wN621U_UbVeoJh_ww8Xmcv47Iz28Bq7ydfdGBzMt8tk/edit?usp=sharing).
Have you considered writing it online? It would be easier to build in Google Doc and much easier to review, annotate and complete.

> There is something I do not understand.
> In the "Channel" menu there is now a "Current" field.
> What does that do?
> Why did you change it?

It's an option to convert and display an electrical current instead of an electrical voltage.
I took this feature from the VDS3104 model.
It's meant to be used with a current clamp or with a probe through a resistor.

> There were some issues here where you added some functions and explained them.
> If it is necessary, we could add another chapter before "Technical Specifications" where different things are [...]

Just add a mention in your doc in red and I'll fill the gaps in English when I'll review it, that is if that's ok with you.

> When I translated the manual I was looking for a very good translator who could handle the technical texts.
> I can recommend www.DeepL.com/Translator (free version) https://www.deepl.com/translator. [...]

Good to know, I'll use this website from now on.

km-linux, 2021-03-04:
> Have you considered writing it online? It would be easier to build in Google Doc and much easier to review, annotate and complete.

For writing my books I prefer doing it on my PC.
Some years ago in the company where I made my money, they switched the Windows 10 on the IBM host. All employees must work with sessions on their clients.
I had to do often to copy & paste from one application to the other. Doing this we had big problems with copying as all were only on sessions. The result was that they installed all needed programs on my PC, that I could do my job.

That's why I'll do it locally and upload an English copy to Google doc.
You can make your changes in red there, which I will copy to my English version and transfer to the German one. 
When everything is done, I'll upload it completely again.

When I'm ready with my changes, I will do the upload for you.

km-linux, 2021-03-05:
> Have you considered writing it online? It would be easier to build in Google Doc and much easier to review, annotate and complete.

I have taken a look at **Google docs** and tested it. 
**It is not usable for us, because it has far too little functionality.** It can do almost nothing with tables, for example.

A solution for the problem is **ONLYOFFICE** https://www.onlyoffice.com/
It is a **browser operated OpenOffice** and can be installed on a **Nextcloud** server as community app.
I tested it on my Nextcloud server at home and it works fine. 

If you have got a Nextcloud server that we could use for this, it would be good. 
Otherwise, there is still the possibility that I install a Nextcloud server with ONLYOFFICE for documentation at my hosting provider.

florentbr, 2021-03-05:
> I have taken a look at Google docs and tested it.
> It is not usable for us, because it has far too little functionality. It can do almost nothing with tables, for example.
>
> A solution for the problem [...]

The tool here is not important. It was just a suggestion and I'd rather keep it simple. Just place your document in a shared folder on Google drive or equivalent.

As a side note, I prefer Google Doc because of the simplicity and I don't find anything lacking. With the heavy weights like Office, I often spend more time searching, tweaking and adjusting than creating content. If it wasn't for the composed screenshots/arrows, I would have suggested text markup. I don't understand how you can do almost nothing with tables since it's possible to create and tweak one down to a single border. Since I managed to create the same table as the original doc in a few seconds, I wouldn't say that it's unusable.

dbadrian, 2021-03-10:
First of all, thanks @km-linux  for writing those great manuals!

I know, its probably a bit annoying to hear another opinion/suggestion for "where\how to write", but have you considered using latex?
I guess, if you never used it, it seems daunting or annoying (frankly it can be), but it is very convenient as the content is written in a plain text file (with images separately stored). The pdf generated from it, however, can be beautifully typeset. The big advantage we get is that you could version control it in a repository here on github, and people can make comment/pull requests. This is how I collaborate with colleagues when writing scientific papers and it works quite well (and the results just look better). Also solves the annoying decentralized problem you described.

Many templates for good looking technical manuals already exist, so its more about filling in the content.

km-linux, 2021-03-11:
I will place the documents in a shared folder on Google drive. The book is nearly ready and I will not start again. 
It's easy to create PDFs from LibreOffice. That is what we need. 

I think it's easier for someone who wants to translate the book to work with LibreOffice, as so many people do not know Latex, me too.

dbadrian, 2021-03-11:
I totally understand, I guess if someone wants to take up that hat in the future, the license does permit him/her to do so.
Thanks again for the excellent manual

km-linux, 2021-03-11:
The manual what you have seen is the draft for the last version. The revision with many changes is just finished and now Florent will correct it. Then I will take everything and incorporate it into the English and German books. 
When it is published, have a look!

About the license:
Everything should be reciprocal, I get something and give something too.

km-linux, 2022-01-09:
Can be closed.
There is a documentation #32 for release cf12 and #43 for release cf13.

closed 2022-01-09 by km-linux