florentbr/OWON-VDS1022 #20: Display Scales on Axis

QSRH1, 2021-02-09:
Is it possible to add scale information to the axes ?; so a display that is more as you would print a page than a traditional oscilloscope.
Clearly, this would be particularly useful in the print mode - but it would also make the main display easier to read.
Thank you for your attention, and even more so for creating this in the first place !

florentbr, 2021-02-09:
The vertical scale in displayed on the bottom/left corner while the horizontal scale is displayed in the bottom right corner next to M. Each brand displays the horizontal/vertical scale differently. So I'm not sure what you want, especially since traditional oscilloscopes don't display information other than a wave. Can you show us an example?

QSRH1, 2021-02-09:
[image]
This is a plot from my program using an HP8903B analyser.
The important bit for this conversation is the numbers, although it would be good to be able to add a descriptive when printing.
NB: I say this as someone who has used (and still uses) traditional scopes for the best part of 50 years - this is one thing that would make viewing and analysing waveforms  significantly easier if it was possible; no more checking the controls, working out where zero is, and then counting the divisions....
Again, thanks for listening

florentbr, 2021-02-09:
I see now where you are going. Adding a scale on the graph would require a significant rework of the interface, especially since each input can have a different scale. While I don't think it's necessary in the main window, I can see the added value in the print preview. 

Unfortunately, I'm not planing on spending time for now on this kind of feature.

Notes that there're mark cursors to quickly measure each axis. If your goal is to print a polished graph, then export the data as CSV and build a graph with a tool like Excel, Matlab, Octave, Jupyter Notebook ...

QSRH1, 2021-02-09:
OK, Thanks for taking the time to consider it.

florentbr, 2021-12-26:
Closing as this feature won't be implemented.
Note that the latest release include an API to directly communicate with the device.
So as an alternative you could plot in a [jupyter notebook](https://github.com/florentbr/OWON-VDS1022#api).

closed 2021-12-26 by florentbr