florentbr/OWON-VDS1022 #21: 20:1 attenuation setting

User-McUserface, 2021-02-15:
I see this oscilloscope slowly but surely gaining popularity, one of the segments being automotive hobbyists and in that field 20:1 attenuation is pretty much standard.

That specific setting in the software would be great, but even better than that, would be support for custom ratios.

I have no idea how easy or hard any of that might be to implement, but I thought I'd put the suggestion out there.

florentbr, 2021-02-15:
Indeed, it would be useful to have the x20 ratio since this attenuator is common. I'll add the ratio since it's just a matter of adding the value in a config file. 

For the record, the list is present in `owon-vds-tiny-1.0.33-cf11.jar` in `com\owon\uppersoft\dso\model\machine\params\VDS1022ONE.txt`:

```
[ProbeRate]
1, 10, 100, 1000
```

User-McUserface, 2021-02-15:
> For the record, the list is present in `owon-vds-tiny-1.0.33-cf11.jar` in `com\owon\uppersoft\dso\model\machine\params\VDS1022ONE.txt`

Thank you for mentioning that, I'm sure it will come in handy.

florentbr, 2021-03-02:
added in release 1.0.33-cf12

closed 2021-03-02 by florentbr