florentbr/OWON-VDS1022 #22: Trigger at >100ms/div

csdpm, 2021-02-15:
This is one that has been a perennial issue with the VDS1022 since day one...

Is there any way to implement trigger capability on timebase settings slower that 100ms/div (50ms/div on the original Owon S/W)?

I know that the display is in roll mode at this point, but it would be great if it could trigger (maybe at a fixed 50% of buffer if it's easier) and roll to the end of the buffer before stopping.

I remember asking Owon about this several times but they never got past saying that they would look at it.

This leaves me wondering if there is some particular difficulty (H/W?) in implementing it.

Many thanks.

florentbr, 2021-02-16:
I've been investigating this feature for a while. One quick workaround is to change the timebase for the rolling/slow move mode. It seems to work up to a point. The downside is that the wave is displayed only at the end of the acquisition. The `[SlowMoveTimebase]` setting is in `owon-vds-tiny-1.0.33-cf11.jar` in `com\owon\uppersoft\dso\model\machine\params\VDS1022ONE.txt`. I've already increased this value from 100ms to 200ms in a previous release.

The difficulty here is the lack of documentation and transparency of the code. I recently found a way to get the cursor position while pulling the buffer in a loop from the device. So it's possible to stitch the frames to build one aligned on a trigger position. But the amount of work required to implement it is quite consequent. That said, it may be possible to simply stop if the triggered position is available in the payload while in roll mode. If not, we'll have to look for the triggered position by software instead of hardware.

csdpm, 2021-02-16:
Many thanks Florent,

I suspected that might be the case. It does sound like an awful lot of guesswork is needed. It's ironic that it might require implementing S/W triggering, when the advantage of the VDS1022 against its competitors is the inclusion of triggering in H/W.

The significance of your change to the roll mode setting hadn't occurred to me until I realized that it allowed triggering down to lower timebase speeds. Thanks for the pointer to where it is set, I think that's enough to allow me to play around with the optimal changeover point for me - I hadn't realized that it could be changed with such a simple setting.

florentbr, 2021-03-02:
The single and normal trigger is now available in release 1.0.33-cf12.

For the record, the trigger position is not present in the payload but there's a command to ask the device the status. I just added a timer to stop the acquisition once the device returns a triggered state. So the triggering is still handled by hardware. The accuracy of the horizontal position on the cursor is more or less the refresh rate which is about 30ms. Given that the rolling mode starts at 100ms/div, this error shouldn't matter.

closed 2021-03-02 by florentbr

csdpm, 2021-03-03:
Hi Florent,

Many thanks for the cf12 update - one problem though, I can't get it to trigger at >100ms/div. I must be doing something stupid, but I set 100ms/div (tried slower too), single trigger (tried normal too) but no trigger. I've also tried both channels.

For trigger source, I an just setting the sensitivity to 50ms/div (x10), trigger level at 100mV and touching the probe tip to the 1kHz cal output. This gives a nice big visible edge waveform, but the roll doesn't stop.

One thing I did notice... When I first started the s/w (w7 64pro), there were two horizontal cursors already set on CH1 (+4V and - 4V), and Channel 1 and 2 Vrms measurement already set up too. Presumably from your last test, but it made me wonder if you had maybe taken the wrong snapshot for the install files.

I can't think what I am doing wrong.

Many thanks,
Chris

csdpm, 2021-03-03:
Florent,

Sorry, I think that may have been finger trouble on my part - The run green arrow top right takes the trigger back to Auto trigger mode (I can't remember if it always did that. If I use the single trigger setting from the trigger menu, or the single trigger icon (top right) it works fine.

Sorry to worry you. Thanks very much!
Chris

florentbr, 2021-03-03:
Right, I forgot the trigger in the dock menu. It doesn't work when set in the dock, only by shortcut or icon.

Single trigger: Enter Key or Single Icon
Normal trigger: N Key

Running the acquisition with the PLAY icon or the SPACE key will switch back to Auto mode.

The single trigger starts the acquisition right away and stops it on the vertical cursor once it's triggered.

The Normal trigger displays the acquisition only when triggered. It's not related to the vertical cursor. The trigged state remains on for 650ms after the last triggered input. This delay is not compressible since it's produced by the device.

Thank you for the feedback and donation.

csdpm, 2021-03-03:
Ah, yes, that makes sense now. Thanks for the clarifications.

I don't see the normal trigger display as being an issue - Single trigger is really the one of most interest.

florentbr, 2021-03-04:
The normal trigger is useful when you are missing a third hand.