florentbr/OWON-VDS1022 #23: Software crashing on Linux

User-McUserface, 2021-02-17:
Tried installing software on my Manjaro KDE desktop.

Everything installed and launched alright but the graph refresh rate seemed very low.

When I pressed ctrl+enter (auto setting), the software struggle for a bit and stopped with message "WriteContent Err and DisConnect".

florentbr, 2021-02-17:
Could be that there's not enough amps on the USB port or too much noise. Try to connect it to an external power supply like a smartphone charger.

I experienced this kind of failure once. It was because the USB port couldn't be accessed in time due to high CPU usage from another application. So make sure that there's nothing else monopolising the resources or ports.

I've been using this scope on a 15 years old compaq nx7000 without any issues so far.

Could you post the exception ? To view the exception, simply run `owon-vds-tiny` from a console.

User-McUserface, 2021-02-17:
> Could you post the exception?

Sure, here it is. This is all that happened from the moment I pressed ctrl+enter to the moment I closed the window after all the prompts.

`wrEmpty
trg set 16 bits:
[toBytesString: ]00000000 00000000
[trg holdoff arg[9~0]1, index[2~0]:]2
arg[9~2] >> 0x0
arg[1~0] 000 index[2~0] >> 0x42
[Send: ] 0xb, 1, 0x3  @CHL_ON_ADD
, 0xb 0x0 0x0 0x0 0x1 , 0x3 
[Response: ] resNum: 5, value: 1
, 0x53 0x1 0x0 0x0 0x0 
[Send: ] 0x10c, 1, 0x0  @EMPTY_ADD
, 0xc 0x1 0x0 0x0 0x1 , 0x0 
[Response: ] resNum: 5, value: 1314212940
, 0x53 0x4c 0x4c 0x55 0x4e 
[Send: ] 0x111, 1, 0xffffffa0  @channel_ch1_ADD
, 0x11 0x1 0x0 0x0 0x1 , 0xa0 
[Response: ] resNum: 5, value: 3
, 0x53 0x3 0x0 0x0 0x0 
[Send: ] 0x116, 2, 0x229  @volt_gain_ch1_ADD
, 0x16 0x1 0x0 0x0 0x2 , 0x29 0x2 
[Response: ] resNum: 5, value: 3
, 0x53 0x3 0x0 0x0 0x0 
[Send: ] 0x10a, 2, 0x1e4  @zero_off_ch1_ADD
, 0xa 0x1 0x0 0x0 0x2 , 0xe4 0x1 
[Response: ] resNum: 5, value: 3
, 0x53 0x3 0x0 0x0 0x0 
[Send: ] 0x24, 2, 0x0  @TRG_ADD
, 0x24 0x0 0x0 0x0 0x2 , 0x0 0x0 
[Response: ] resNum: 5, value: 1
, 0x53 0x1 0x0 0x0 0x0 
[Send: ] 0x26, 1, 0x0  @trg_holdoff_arg_ch1_ADD
, 0x26 0x0 0x0 0x0 0x1 , 0x0 
[Response: ] resNum: 5, value: 1
, 0x53 0x1 0x0 0x0 0x0 
[Send: ] 0x27, 1, 0x42  @trg_holdoff_index_ch1_ADD
, 0x27 0x0 0x0 0x0 0x1 , 0x42 
[Response: ] resNum: 5, value: 1
, 0x53 0x1 0x0 0x0 0x0 
[Send: ] 0x2e, 2, 0x444e  @[edge_level_ch1_ADD] @0x2e, 2 bytes
, 0x2e 0x0 0x0 0x0 0x2 , 0x4e 0x44 
[Response: ] resNum: 5, value: 1
, 0x53 0x1 0x0 0x0 0x0 
wrEmpty
trg set 16 bits:
[toBytesString: ]00000000 00000000
[trg holdoff arg[9~0]1, index[2~0]:]2
arg[9~2] >> 0x0
arg[1~0] 000 index[2~0] >> 0x42
trg set 16 bits:
[toBytesString: ]10000000 00000000
[trg holdoff arg[9~0]1, index[2~0]:]2
arg[9~2] >> 0x0
arg[1~0] 000 index[2~0] >> 0x42
wrEmpty
trg set 16 bits:
[toBytesString: ]11000000 00000000
[trg holdoff arg[9~0]1, index[2~0]:]2
arg[9~2] >> 0x0
arg[1~0] 000 index[2~0] >> 0x42
wrEmpty
trg set 16 bits:
[toBytesString: ]10000000 00000000
[trg holdoff arg[9~0]1, index[2~0]:]2
arg[9~2] >> 0x0
arg[1~0] 000 index[2~0] >> 0x42
wrEmpty
[... 118 lines cut ...]
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
sleep 50 ms on fail
onFail: Wed Feb 17 19:25:50 GET 2021
tryRescue...
tryRescue fail
sleep 50 ms on fail
windowClosing
`

florentbr, 2021-02-17:
The exception indicates that the USB connexion was lost. It's also possible that the device felt in an unexpected state due to a bad calibration.

To pinpoint the issue :
Try a different power supply and USB cable to exclude an hardware issue
Try to run the device on a light live usb like MX linux to exclude an OS issue
Try each volt base on both channels with a DC coupling  and look for an outstanding value in the calibration (F2)
Run Autoset multiple time and check if it occurs randomly or always after sending :
  `@[edge_level_ch2_ADD] @0x30, 2 bytes , 0x30 0x0 0x0 0x0 0x2 , 0x32 0x28 `

User-McUserface, 2021-02-17:
> To pinpoint the issue

I tried almost everything you suggested. Even booted into a Win partition I haven't used in months. Everything worked perfectly there connected exactly like on linux.

Long story short, on linux it works but only when connected to one particular USB port (out of 4 exactly similar ones). Who knows why, but what I do know is that I'm not qualified to find out :D

Thank you for all your suggestions and prompt replies :+1:

florentbr, 2021-02-17:
Well, if it works fine on the same USB port on Windows, either Linux handles the power differently or it's a driver issue. Check the logs for a clue with `journalctl --since "10 min ago"`. 

You could try to use a different libusb driver. Run
```
sudo rm -f /usr/lib/owon-vds-tiny/libusb-0.1.so.4
sudo pacman -S libusb-compat
```
It will force the app to use the libusb driver from the Arch repo. Then relaunch the app to see if there's a difference. To switch back, just reinstall the app.

User-McUserface, 2021-02-18:
I dug around some more in the motherboard manual and confirmed that the port is indeed high power (3A in fact), but also that it's USB 3.1 and controlled by an ASMedia controller.

I think it's safe-ish to assume that there's something fishy going on with drivers for that controller on linux, while on windows everything is dandy because of the manufacturer supplied drivers.

closed 2021-02-18 by User-McUserface

florentbr, 2021-02-18:
It looks like it's an issue between libusb and the ASMedia controller : 
https://github.com/libusb/libusb/search?q=Asmedia&type=issues

It's probably been been fixed in libusb 1.0. Unfortunately this app rely on libusb 0.1 which is no longer maintained and has a different API.

florentbr, 2021-03-02:
I switched the driver to [libusb-compat](https://github.com/libusb/libusb-compat-0.1) which allows applications written for libusb-0.1 to work
with libusb-1.0. The release 1.0.33-cf12 should now work with your ASMedia controller.