florentbr/OWON-VDS1022 #24: Interface not detected on linux after software restart

User-McUserface, 2021-02-18:
Basically when I have everything up and running and from the upper left corner I disconnect from the scope, software still recognizes that it's there and allows me to reconnect to it, but if I then close the software completely and restart it, it no longer recognizes that the scope is connected until I physically disconnect and reconnect it.

florentbr, 2021-02-18:
What is the status of the led on the scope before closing the app and after reopening ? Do you get an exception in the console before closing the app and after? My device reconnects automatically as soon as I reopen the app on either Windows or Linux. Since I have an Intel USB controller, I guess it's related to your USB ASMedia controller. Check in your bios if there are settings related to your USB controller.

User-McUserface, 2021-02-18:
> What is the status of the led on the scope before closing the app and after reopening?

It's green with occasional red flashes before closing and just green after that, including when app is reopened.

> Do you get an exception in the console before closing the app and after?

Nope, no exceptions at any point.

> I guess it's related to your USB ASMedia controller.

Since I figured out which ports were controlled by ASMedia controller, I started using the ones which are controlled by Intel controller and everything's been working fine, except for this issue with restarting.

km-linux, 2021-02-18:
The oscilloscope has an USB 1.1 port.  **You MUST connect it to an USB 2.0 port.** 
On USB 3.0 port you will have no chance for running.

User-McUserface, 2021-02-18:
> On USB 3.0 port you will have no chance for running.

It does though, I just got done prototyping for today and it's been working through USB 3.0 port all this time. Since my motherboard doesn't have any USB 2.0 ports.

In fact it works in both, Windows 10 and Linux.

florentbr, 2021-03-02:
It should be fixed in release 1.0.33-cf12.
Feel free to reopen this ticket if it's not the case.

User-McUserface, 2021-03-03:
Thanks for keeping up with improvements, I really appreciate it.

As for this issues specifically, I tried the new build and it's still the same.

I don't seem to be able to reopen the issue so I hope you still get the notification.

florentbr, 2021-03-03:
I can't reproduce your issue. Can you at least provide the full logs from the first and second launch?

Run `owon-vds-tiny &>> launch1.log`
Disconnect and close the software
Run `owon-vds-tiny &>> launch2.log`
Post the files launch1.log and launch2.log

User-McUserface, 2021-03-03:
> Can you at least provide the full logs from the first and second launch?

I've done everything else you've requested so far have't I? :)

Here you go.

During launch1 I connected and disconnected the scope from software button, then started launch2 session ad as expected, it was not detected by software.

[launch1.log](https://github.com/florentbr/OWON-VDS1022/files/6077758/launch1.log)
[launch2.log](https://github.com/florentbr/OWON-VDS1022/files/6077759/launch2.log)

florentbr, 2021-03-03:
> I've done everything else you've requested so far have't I? :)

Sorry my native language is not English, is my writing offensive somehow ?

User-McUserface, 2021-03-04:
> Sorry my native language is not English, is my writing offensive somehow?

Not offesive, don't worry about it, it's just that inclusion of "at least" can be interpreted to imply an unsaid part, like "if not X, then at least Y". All I said was that there was no X that I didn't provide for there to be "Y that I could at least provide" :)

florentbr, 2021-03-04:
> Not offesive, don't worry about it, it's just that inclusion of "at least" can be interpreted to imply an unsaid part, like "if not X, then at least Y". All I said was that there was no X that I didn't provide for there to be "Y that I could at least provide" :)

Thank you for the explanation, it wasn't my intention to imply anything.

> During launch1 I connected and disconnected the scope from software button, then started launch2 session and as expected, it was not detected by software.
> launch1.log
> launch2.log

I fixed the Exception present in both of the logs but it may not be the source of your issue.
Can you try this release :

[OWON-VDS1022-1.0.33-cf12.1.zip](https://drive.google.com/file/d/18R5FHr7Q3HiWHPOTWYQl0a1X-ZUIrkzi/view?usp=sharing)

User-McUserface, 2021-03-04:
> Can you try this release

I tried and the issue is still present. To simplify it further: if the device is plugged in first and software is launched later, it won't be detected. If it's done vice-versa, first software and then device plugged in, it detects and works just fine.

I logged two launches again just in case. Scope was plugged in before launch1 and during launch2.

[launch1.log](https://github.com/florentbr/OWON-VDS1022/files/6085505/launch1.log)
[launch2.log](https://github.com/florentbr/OWON-VDS1022/files/6085506/launch2.log)

florentbr, 2021-03-04:
I don't think this software is the issue here. I've refactored all the code related to libusb. So I'm confident that it's something else. It looks like libusb simply fails to enumerate the ports for some reasons.

If you want to get to the bottom of it, you'll have to dig into the logs and monitor the USB ports. I'd start with `lsusb` to see if the device is also missing once relaunched.

I'll try to download a live version of Manjaro KDE this weekend just in case it's related to your distro.

User-McUserface, 2021-03-04:
> I'll try to download a live version of Manjaro KDE this weekend just in case it's related to your distro.

Honestly, it's not such a big problem to warrant a big time investment from you.

When I open an issue, my aim is to bring a problem to the attention in case it's something that might affect significant amount of people.

You took a look at the code and everything seemed in order, plus nobody else seems to be having the same issue.

Of course it's up to you, but I'd rather you concentrate on other issues or improvements which more people could benefit from.

florentbr, 2021-03-04:
I agree. I'm not planing to investigate this issue any further.
I just wanted to take this opportunity to see how KDE has evolved.

BorisDess, 2021-06-09:
Hello guys. I'm a Macbook M1 user.
I've been able to install and run the app using rosetta and Azul ARM Java build.
However I have the same issue as described above. The USB/Serial seems to not work.
Led is red with green every 3 secs.

florentbr, 2021-06-09:
@BorisDess, your issue is not the same if your USB/Serial doesn't work. Could you please move your new issue to a new one? thanks.

florentbr, 2021-12-26:
It should be fixed in [release 1.1.1-cf13](https://github.com/florentbr/OWON-VDS1022/tags)
Feel free to reopen this issue if it's not the case.

closed 2021-12-26 by florentbr