florentbr/OWON-VDS1022 #26: serial decoder

John-By, 2021-02-28:
One major limitation of most or even all Software, available for the VDS1022(i), is the absence of serial bus decoders.
Those are a useful feature for checking signal integrity.

Therefore, I want to ask, if it possible to add some, supporting for example UART or I2C?

SilvainLouis, 2021-03-03:
Hi,
This is a feature that I would also be interested in.
Wouldn't it be too complicated to implement in java in your project ?
Would you have some starting point to help me to program this ?
Thanks,

florentbr, 2021-03-03:
This software was not designed to continuously pulls the samples from this device.
It pulls a 5k buffer per channel every 30ms or so and the data is already aligned on the triggered position. So assuming 6 samples per clock period, it would provide at best 833 clock periods of raw data.

I'm not sure it's worth the effort since the starting price for a logic analyser is 10$.

That said, if someone is willing to provide a Java class to decode a buffer, I can try to add a decoder.

Another option would be to implement a plug-in for [sigrok](https://sigrok.org/).
I think it's possible since there's a way to continuously pull the buffer and aggregate the frames.
Considering the top speed of USB2, the top signal frequency should be around 1Mhz.

SilvainLouis, 2021-03-03:
Yes, if it's possible to link your software with a plug-in for sigrok, it's a very good option.
Sigrok have a api to send data to the decoder.
Any idea to make it ?

florentbr, 2021-03-03:
> Yes, if it's possible to link your software with a plug-in for sigrok, it's a very good option.
> Sigrok have a api to send data to the decoder.
> Any idea to make it ?

Someone already started to implement a plugging for sigrok:
https://github.com/noopwafel/libsigrok/blob/master/src/hardware/owon-vds-tiny/protocol.c#L275

For the protocol, have a look here:
https://github.com/albert-spruyt/pyVds1022