florentbr/OWON-VDS1022 #28: Error Mark Cursor Menu: Aplitude no color change

km-linux, 2021-03-04:
Hi Florent,
While testing for the new documentation, I noticed an error in the 
**Mark Cursor Menu:**
**Channel**
**Amplitude** color always stays green, when selecting an other color.

[image]

florentbr, 2021-03-04:
As of release cf12, the amplitude cursor takes the colour and measures of the selected/highlighted channel. Switching between channels with TAB or mouse will change the colour and values accordingly. The colour button in Cursor/Channel now only affects the time cursor. I haven't touched yet the cursors of the FFT graph.

km-linux, 2021-03-04:
When I have changed the documentation, I will close the issue.

km-linux, 2021-03-06:
Documentation changed.

km-linux, 2021-03-07:
> Switching between channels with TAB or **mouse** will change the colour and values accordingly.

**How to switch with the mouse?**
The **pulldown menu** for channel selection **behind Channel --> Amplitude does not exist** in this release.

florentbr, 2021-03-07:
> How to switch with the mouse?
> The pulldown menu for channel selection behind Channel --> Amplitude does not exist in this release.

To select a channel, click on the channel area on the main window (element 18/19 from the doc). The selected channel is also used by some shortcuts like invert, coupling, offset, trigger, probe ratio.

km-linux, 2021-03-12:
Fixed

closed 2021-03-12 by km-linux