florentbr/OWON-VDS1022 #29: Russian language support

LeonIndman, 2021-03-10:
Hello florentbr! Thank you very much for another successful software update!
I would like to make corrections to the Russian language support files. I am attaching them to my post.

LeonIndman, 2021-03-10:
[image]
[shortcuts.zip](https://github.com/florentbr/OWON-VDS1022/files/6115512/shortcuts.zip)
[MsgLib_ru.zip](https://github.com/florentbr/OWON-VDS1022/files/6115547/MsgLib_ru.zip)

LeonIndman, 2021-03-10:
I would also like to ask you to correct a small blot in this menu, which is shown in the 
[image]
screenshot.

km-linux, 2021-03-11:
Hello  LeonIndman,
I'm writing a completely new book for the software.
look at issue **German documentation for Owon VDS1022(I)/VDS2052(I)**

**If you are interested for translation in Russian language let me know.**

I write it with **OpenOffice, text and images separated.** 
The book is nearly ready.
It will be corrected by florentbr.
I write the book in two languages English and German.
I will place the documents in a shared folder on Google drive.

LeonIndman, 2021-03-11:
Hello km-linux!
Thanks for the offer! When the book is ready, I will be able to see how extensive the work of translation into Russian will be.

florentbr, 2021-12-26:
Fixed in release [1.1.1-cf13](https://github.com/florentbr/OWON-VDS1022/tags)

closed 2021-12-26 by florentbr