florentbr/OWON-VDS1022 #35: WriteContent Err and DisConnect

xtradark812, 2021-08-07:
When i plug in the device, it is recognized by the software, then it attempts to initialize and then it fails, shows the "WriteContent Err and DisConnect" error, then attempts to initialize again, continuously. Here's the debug (running on archlinux) https://pastebin.com/QhPiKwxG . I attempted to connect it to my windows PC also and received the same error. I hope its not a faulty device.

florentbr, 2021-08-07:
It sure looks like an hardware issue. From the logs I can see that the USB connection is lost while sending the firmware and the device no longer reply. It could be a power issue. Try to add a smartphone charger on the cable or try a different cable.
Try multiple times and check the logs. If it fails randomly, then it's likely a cable issue or a bad soldered joint on one of the components. If it always fails at the same step, then there's something wrong on the board. I would inspect the board and try to resolder the joints.

xtradark812, 2021-08-07:
> It sure looks like an hardware issue. From the logs I can see that the USB connection is lost while sending the firmware and the device no longer reply. It could be a power issue. Try to add a [...]

I think you are correct, it is a hardware issue. I also tried switching cables and using a phone charger but nothing worked. I will return the device tomorrow morning.

closed 2021-08-13 by florentbr

florentbr, 2021-08-24:
By the way, did you try with the official release from OWON ? I just noticed that there's a new release 1.0.34 which has now two different firmware (FPGA). The firmware is now loaded according to the version of the board. Your issue could be explained if you had the new version of the board while runing the customized release 1.0.33-cf12 .