florentbr/OWON-VDS1022 #38: New FPGA config file version

csdpm, 2021-10-02:
Hi Florent,

Just to make you aware that there is a new version of the FPGA config file (V3.9) which apparently improves pulse triggering.

Forum link...  https://www.eevblog.com/forum/testgear/owon-vds1022i-quick-teardown-(versus-the-hantek-6022be)/msg3723262/#msg3723262

florentbr, 2021-10-02:
Hi, I'm aware. I've already merged version 1.0.36. The app now chooses between 3 different FPGA depending on the serial of the device.
* **V2.7.0** : VDS1022_FPGAV3_*.bin
* **V2.4.623, V2.6.0** : VDS1022_FPGAV2_*.bin
* **otherwise** : VDS1022_FPGAV1_*.bin

The issue is that with release v1.0.36 the pulse/slope trigger no longer work with my device (V2.5). It's sending new commands to the device to define the width of the pulse. I don't know if it's related to my version or if it's just poor coding/testing from OWON. It would be thefirst time. I'll keep the old commands in the next release if I can't fix these changes.

csdpm, 2021-10-03:
Good to know that you're already on top of it.

Thanks for insight- good old Owon! :D

florentbr, 2021-12-26:
I added the new firmwares in [release 1.1.1-cf13](https://github.com/florentbr/OWON-VDS1022/tags)
The app now loads the firmware according to the board version.
For the record, they restored the old commands in release 1.1.1 for the triggers when an older board is detected.

closed 2021-12-26 by florentbr

csdpm, 2022-04-16:
Hi Florent,

Just to let you know. The latest version of the Owon S/W on their website now contains five FPGA firmware images! I just checked your V1.1.1-cf15 and noticed that it only contains four (still one more than I expected!).

The following thread may possibly be relevant... https://www.eevblog.com/forum/testgear/are-you-tired-of-being-treated-like-an-idiot/

florentbr, 2022-04-17:
Thanks for the info. Looks like they are still struggling to get those FPGA chips.
For the record, the firmware is loaded according to the board version:

   * V5 : VDS1022_FPGAV5_*.bin
   * V4 : VDS1022_FPGAV4_*.bin
   * V3, V2.7.0 : VDS1022_FPGAV3_*.bin
   * V2.4.623, V2.6.0 : VDS1022_FPGAV2_*.bin
   * otherwise : VDS1022_FPGAV1_*.bin