florentbr/OWON-VDS1022 #40: Probe Rates

mikerocklewitz, 2021-11-19:
Would it be possible to add x50 and x500 probe rates to the app? Would make life a billion times better with differential probes :D

florentbr, 2021-11-19:
It's quite easy to add a probe rate:
1 - Open `owon-vds-tiny-1.0.36-cf13.jar` with 7zip or equivalent.
2 - Extract `com\owon\uppersoft\dso\model\machine\params\VDS1022ONE.txt`
3 - Edit the line under `[ProbeRate]` with the desired values
4 - Overwrite `VDS1022ONE.txt`

mikerocklewitz, 2021-11-19:
Well that was easier than i thought! Thank you!

[image]

closed 2021-11-19 by mikerocklewitz

florentbr, 2021-12-26:
Included in [release 1.1.1-cf13](https://github.com/florentbr/OWON-VDS1022/tags)

mikerocklewitz, 2021-12-26:
> Included in [release 1.1.1-cf13](https://github.com/florentbr/OWON-VDS1022/tags)

I had emailed owon about this once.  It literally took 3 months to get a response saying they would tell their "engineers".

I thought that was amusing thanks for everything you do!  Have a good new year!

florentbr, 2021-12-26:
> I had emailed owon about this once. It literally took 3 months to get a response saying they would tell their "engineers".

You are lucky, they never replied to my emails.

> I thought that was amusing thanks for everything you do! Have a good new year!

Thanks and same to you!