florentbr/OWON-VDS1022 #41: OWON VDS1022I not detected on Ubuntu 20.04

ReportX, 2021-12-19:
It works fine in an old Windows XP laptop (OWON VDS C2 version 1.1.1) but my aim is to get it working on Ubuntu 20.04.

On Ubuntu 20.04 it looks the scope is detected,

$ lsusb
Bus 001 Device 019: ID 5345:1234 Owon PDS6062T Oscilloscope

but when I launch the application, its try to iniatilize but finally it is unable to find the hardware.

Clicking on the top-left panel (Auto/Syncing/...) got a drop-down menu which an "Connect to USB" option and "VS1022, bcd: 0x200". After select this port the results are the same, Initializing, later Syncing and finally "WriteContent Err And DisConnect [X]"

It seems that the program is able to find the USB port because after disconnect and connect the USB cable the top-left panel shows "USBFound" and the drop-down menu to select "VS1022, bcd: 0x200".

Some other tests and results:

$ owon-vds-tiny &>> launch1.log

after close the application, disconnect and reconnect the scope:

$ owon-vds-tiny &>> launch2.log

after close the application and simply disconnect the scope (keeping it offline):

$ owon-vds-tiny &>> launch3.log

I see an exception (launch1.log) and messages saying something about download some FPGA (firmware?) but I don't really know how to manage it. Any help would be appreciated.

Thanks in advance
[launch1.log](https://github.com/florentbr/OWON-VDS1022/files/7740702/launch1.log)
[launch2.log](https://github.com/florentbr/OWON-VDS1022/files/7740703/launch2.log)
[launch3.log](https://github.com/florentbr/OWON-VDS1022/files/7740704/launch3.log)
.

florentbr, 2021-12-19:
OWON recently released different firmwares for different boards, likely due to the chip shortage.
Your device has a V2.6.0 board. It needs to load the firmware `VDS1022_FPGAV2_V4.1.bin` which is not provided with the current release. So you can either wait for the next release or you can try to replace the firmware manually with the correct one included in `OWON_VDS_C2_1.1.1_Setup.exe`.

ReportX, 2021-12-19:
Thank you very much for your quick reply. I'm very interested in upgrade manually the firmware even using the Windows laptop.

"About" at the Utility menu says my version is V2.6.0 but when  clicking Update button the application tries to resolve www.owon.ca which doesn't exist so later says "Cannot update from the Internet,... please download the latest software from web site." As you know, there isn't a new version on their web site.

I have found any firmware in the OWON_VDS_C2_1.1.1_Setup.exe but taking a look in the installation directories, I have found two directories which maybe related,
- 	flash_txt (2 files)
- 		flashmemory.txt
- 		init.txt
and
- 	fpga (2 directories, 5 files)
- 		vds1022 (4 files)
- 			VDS1022_FPGAV1_V3.9.bin
- 			VDS1022_FPGAV2_V4.1.bin
- 			VDS1022_FPGAV3_gaoyun.bin
- 			VDS1022_FPGAV4_gaoyun.bin
- 		vds2052 (1 file)
- 			vds2052_FPGA_v1.0.8.bin
but there is no VDS1022_FPGAV2_V4.1.bin file and I was unable to find any executable capable of perform a firmware upgrade. Am I missing something?

rcalders, 2021-12-19:
@ReportX, I had today the same issue, on a Windows platform, with my newly purchased VD1022iI. I've followed the workaround provided by Florent (big thank you by the way!). I've deleted the whole content of  "C:\Program Files\OwonVdsTiny\fwr" folder and copied inside the VDS1022_FPGAV2_V4.1.bin file (you can find it at "C:\Program Files (x86)\OWON\VDS_C2\fpga\vds1022\VDS1022_FPGAV2_V4.1.bin" (from the original software). After launching everything worked fine.

ReportX, 2021-12-20:
Thank you @florentbr  and @rcalders. Finally it is working fine.
I have just to remove all files on /usr/share/owon-vds-tiny/fwr on the Linux computer then copy to /usr/share/owon-vds-tiny/fwr the VDS1022_FPGAV2_V4.1.bin file copied from the fpga directory of the original OWON software installed in the Windows computer.

florentbr, 2021-12-26:
The new firmwares where added to [release 1.1.1-cf13](https://github.com/florentbr/OWON-VDS1022/tags)
The app now loads the firmware according to the board version.

closed 2021-12-26 by florentbr