florentbr/OWON-VDS1022 #43: Modifications in Manuals for release cf13

km-linux, 2021-12-26:
Hi Florent,
you have got the English and German manuals for release cf12.

Let me know which changes you did.
**I will modify the manuals for cf13.** 
Than you can include them into the new release for download.

Regards

florentbr, 2021-12-28:
It's still in working progress, though it's been a while since the last time I spent time on it.
All the changes for release cf13 are described in changes.txt

km-linux, 2022-01-07:
Hi Florent,
**I changed all affected images** and moved **MUTLI** setting from **Utility** to **Channel** menu.
Menu **Assist Set** is deleted.
I made some changes in **MsgLib_de.properties** for some corrections.

I don't know where to place your description of the **API (Jupyter Notebook)**. 
My first idea was to place it at **II. VDS Oscilloscope Requirements, API & Help**.
What do you think about it?

Here you get a look as draft to the new book:
[VDS_User_Manual_cf13_en.pdf](https://github.com/florentbr/OWON-VDS1022/files/7831102/VDS_User_Manual_cf13_en.pdf)

Please label my issues with **documentation** (Improvements or additions to documentation).
It's easier to find them. Perhaps someone will look into it and write, if something is needed or must be corrected.

km-linux, 2022-01-07:
Hi Florent,
in Google Drive I moved the old documentation to a new sub-directory **cf12**.
https://drive.google.com/drive/folders/1FXqKiYuIB3uqf9DrArqeJI-nh7sOujOK

The new documentation will be saved in the new sub-directory **cf13**.