florentbr/OWON-VDS1022 #44: sec/div centered in dropdown

ericdurfee, 2022-01-02:
I mostly use a mouse. Would be nice if when clicking on time/div, the current setting was in the middle of the dropdown. Then I wouldn't have to scroll to change the sweep to 10x faster.

florentbr, 2022-01-03:
I agree, the Combobox is not very handy. It's a Java component with no direct way to set the scrolling position.
It would require to create a new component which is not a trivial thing.

At some point I thought of adding some gestures on the chart :
Double click/tap left : reduce the timebase
Double click/tap right : increase the timebase

But since I only use shortcuts (+,-), I never implemented it.

Have you tried to scroll the timebase directly without clicking on it?
If I remember correctly, scrolling directly on a setting will set the next/previous value.

ericdurfee, 2022-01-06:
Thanks. Scrolling time base worked great, after I cleaned the mouse wheel.

closed 2022-02-09 by florentbr