florentbr/OWON-VDS1022 #46: New install problem - repeat of previous issue #2

TechLobo, 2022-01-07:
I have just purchased an Owon VDS1022i and installed the software which was downloaded from the Owon site - this works OK in that I can see the USB device including the calibration trace when connected.

I had been reading Gyro's breakdown thread and was interested in trialing the improved interface that you have produced, so downloaded and installed the software from Github.

I appear to be experiencing the same issues as described in a previously closed issue (https://github.com/florentbr/OWON-VDS1022/issues/2).

Initially I had the sharing violation during install due to copying the msv*100.dll's which prevented the software being installed. This was overcome by manually copying in the later dll's. The install then completed successfully.

No mention of switching to 32-bit install was made during the install process, so I assume that the 64-bit version was installed.

The application starts but in Offline mode - if you click on the Offline 'button' it shows 'No available USB found >'

Switching back to the original Owon software shows successful synchronisation and display of the calibration signal.

I am running on Windows 10 Pro (64-bit), and initially had OJDK 17 installed.

As part of fault finding I uninstalled  Java17 and installed Java SE 8, and then subsequently Java 11 - same issue in all cases - application starts but Offline, with no available USB found.

Starting the application in debug mode shows the following:

java.lang.NoClassDefFoundError: Could not initialize class ch.ntb.usb.LibusbJava
        at com.owon.uppersoft.vds.core.usb.CDevice.init(CDevice.java:191)
        at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:207)
        at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
        at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
        at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
        at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:30)
        at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:75)
        at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)

Any ideas on how to proceed?
Thanks.

TechLobo, 2022-01-07:
Got it working but had to install OJDK 8.

Subsequent upgrade to OJDK 11 then also worked, although it hadn't when first installed (without previous install of OJDK 8).

Not that it matters now that I know.

Now just need to investigate the possibilities ;-)

closed 2022-01-07 by TechLobo

florentbr, 2022-01-07:
It works with Java 8, 11 and 17 with both 32 and 64 bits versions.
Running `launch.cmd` in the installed folder would have shown you exactly what the problem was.
The issue you are describing is not related to Java, but with the dependent libraries, likely with the version of the Microsoft C Runtime Library.