florentbr/OWON-VDS1022 #47: Device and/or software freeze @100 mV or 50 mV with signal on screen

rinntimo, 2022-01-14:
If I have both signal more than 100 mV AND I set the scale for 100 mV or less the whole thing freezes.
BUT if I don't have signal or I have AC-coupling it doesn't freeze. OR if the signal reference point is so high up that the signal will not fit on screen at all when going 100 mV scale - it will not freeze!
Might be related to the latest add-on with 100 mV scale?
Does not recover unless both restarting the device (USB cable disconnect and reconnect) and restarting the application.
Using the latest version 1.1.1-cf13
"About" telling Version 4.0.0 
Have tested in both Win10 and Ubuntu.

florentbr, 2022-01-14:
Please post the full logs (run `owon-vds-tiny` in a terminal/console).
Does the issue occur with the original OWON software v1.1.1 ?
Have you tried to calibrate the device ?

rinntimo, 2022-01-14:
`found device idVendor:0x5345 idProduct:0x1234 serialNumber:VDS1022
sn: VDS1022
should FPGA re-download?
FPGA_DOWNLOAD_QUERY_ADD
fpga running: 1
beforeFPGADownload: false
rn: 2002
fetch flash crc32: 592712545
Gain
Step
Compensation
board version: V4.0.0
trg set 16 bits:
[toBytesString: ]00000000 00000000
wrEmpty
wrEmpty
[Send TRG_ADD 0x0] S 0
[Send trg_holdoff_ch1_ADD 0x8002] S 0
[Send edge_level_ch1_ADD 0xb1bb] S 0
[Send channel_ch1_ADD 0xffffffa0] S 0
[Send volt_gain_ch1_ADD 0x224] S 0
[Send zero_off_ch1_ADD 0x79d] S 0
[Send CHL_ON_ADD 0x1] S 0
[Send channel_ch2_ADD 0x20] S 0
[Send volt_gain_ch2_ADD 0x225] S 0
[Send zero_off_ch2_ADD 0x5af] S 0
[Send DM_ADD 0x13ec] S 0
[Send SYNCOUTPUT_ADD 0x0] S 0
[Send TIMEBASE_ADD 0x14] S 0
[Send SAMPLE_ADD 0x0] S 0
[Send SLOWMOVE_ADD 0x0] S 0
[Send PRE_TRG_ADD 0x9f6] S 0
[Send SUF_TRG_ADD 0x9f6] S 0
[Send EMPTY_ADD 0x1] S 4e554c4c
[Send channel_ch1_ADD 0xffffffa0] S 4
[Send volt_gain_ch1_ADD 0x2a7] S 0
[Send zero_off_ch1_ADD 0x79a] S 0
Exception in thread "Thread-0" java.lang.ArrayIndexOutOfBoundsException: Index 256 out of bounds for length 256
	at com.owon.uppersoft.vds.core.measure.VerticalValueMeasure.getVtopVbase(VerticalValueMeasure.java:54)
	at com.owon.uppersoft.vds.core.measure.VerticalValueMeasure.doMeasure(VerticalValueMeasure.java:232)
	at com.owon.uppersoft.vds.core.measure.MeasureADC.doVerticalMeasure(MeasureADC.java:42)
	at com.owon.uppersoft.dso.function.measure.MeasureManager.doWaveFormMeasure(MeasureManager.java:318)
	at com.owon.uppersoft.dso.function.measure.MeasureManager.measure(MeasureManager.java:113)
	at com.owon.uppersoft.dso.global.ControlManager.measure(ControlManager.java:277)
	at com.owon.uppersoft.dso.global.ControlManager.updateMeasure(ControlManager.java:255)
	at com.owon.uppersoft.dso.wf.common.MultiReceiver.receiveRTData(MultiReceiver.java:204)
	at com.owon.uppersoft.dso.global.DataHouse.receiveRTData(DataHouse.java:284)
	at com.owon.uppersoft.vds.source.comm.data.GetDataRunner2.getData(GetDataRunner2.java:144)
	at com.owon.uppersoft.dso.source.comm.AbsGetDataRunner.getDataT(AbsGetDataRunner.java:33)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:100)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)`

No, the issue is not there with the original 1.1.1.
Yes I have calibrated the device several times.

rinntimo, 2022-01-15:
> Päivämäärä: 14.1.2022 23.32 (GMT+02:00)
> Saaja: florentbr/OWON-VDS1022 ***@***.***>
> Kopio: rinntimo ***@***.***>, Author ***@***.***>

Is it possible to install this software to a RaspBerry Pi?
I tried to download and install this install-linux.sh to a RP 4B but it says in the beginning that "Architecture not supported: armhf"

Thanks!

-TR

Lähetetty laitteesta Galaxy

-------- Alkuperäinen viesti --------

florentbr, 2022-01-16:
I tracked down the exception. It's related to the measures. It occurs when the voltage offset is not 0 and when the absolute average goes off-screen. I'll fix it in the next release.

Note that there's also an issue with your calibration. Your zero-offset is way off :
Yours : `1949  (0x79a)`
Original: `546 (0x222)`

Calibration example (original software):
+ `C:\Program Files (x86)\OWON\VDS_C2\flash_txt\init.txt`

Calibration  of your device (original software):
+ `C:\Program Files (x86)\OWON\VDS_C2\flash_txt\flashmemory.txt`

With the original software, the calibration can be edited by pressing `Ctrl+L` and `tune` next to the home button.
With this customized version, just press `F2`.

>Is it possible to install this software to a RaspBerry Pi?

Yes it's possible. Just add the following file structure before running the install script :
+ lib/armhf/libusb-0.1.so.4
+ lib/armhf/libusbJava.so

You'll find the binaries in release 1.0.33-cf12:
https://github.com/florentbr/OWON-VDS1022/tree/1.0.33-cf12/lib/linux

You may have to install `libusb-compat`. The source code for `libusbJava.so` :
https://github.com/florentbr/OWON-VDS1022/raw/master/lib/libusbJava-src.zip

florentbr, 2022-02-09:
fixed in release [1.1.1-cf14](https://github.com/florentbr/OWON-VDS1022/tags)

closed 2022-02-09 by florentbr