florentbr/OWON-VDS1022 #48: Playback skipping frame

cgimenez, 2022-01-28:
Hi, dunno if I'm doing something wrong but in an effort to visualize a glitch produced by my DSP code, I recorded 300 frames of signal.
When played back, I can see the glitch passing by on screen but I can't stop/focus on the offending frame using the playback slider. If the offending frame's number is (eg.) 20, I can visualize frame 19 and frame 21 but not the 20.
For now, the only solution I found is setting a very slow playback rate and press space at the right time.

Java 11 / Mac OS Catalina

florentbr, 2022-01-28:
> I can visualize frame 19 and frame 21 but not the 20.

The slider probably just lacks precision when it is dragged due to too many frames. Clicking on the slider and using the Left/Right key should provide a single frame jump.

cgimenez, 2022-01-29:
Thanks a lot ! That's perfect ;-)

closed 2022-01-29 by cgimenez