florentbr/OWON-VDS1022 #50: Offline > No available USB Found "java.lang.NoClassDefFoundError: Could not initialize class ch.ntb.usb.LibusbJava"

piannone, 2022-03-04:
Hello, I have a similar problem to: https://github.com/florentbr/OWON-VDS1022/issues/12

[image]

```
Last login: Thu Mar  3 20:46:46 on ttys001
/Applications/OWON\ VDS1022.app/Contents/MacOS/launch ; exit;                   
piannone@Peters-MBP ~ % /Applications/OWON\ VDS1022.app/Contents/MacOS/launch ; exit;
env: Mac OS X, Java 17.0.2 aarch64
app dir: /Applications/OWON VDS1022.app/Contents/Resources
user dir: /Users/piannone/Library/Application Support/OWON VDS1022
locale: en, en_US, false
java.lang.UnsatisfiedLinkError: Can't load library: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusbJava.dylib
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2393)
	at java.base/java.lang.Runtime.load0(Runtime.java:755)
	at java.base/java.lang.System.load(System.java:1953)
	at ch.ntb.usb.LibusbJava.load(LibusbJava.java:407)
	at ch.ntb.usb.LibusbJava.<clinit>(LibusbJava.java:386)
	at com.owon.uppersoft.vds.core.usb.CDevice.init(CDevice.java:211)
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:227)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
java.lang.NoClassDefFoundError: Could not initialize class ch.ntb.usb.LibusbJava
	at com.owon.uppersoft.vds.core.usb.CDevice.init(CDevice.java:211)
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:227)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
```

piannone, 2022-03-04:
Looks like `/Applications/OWON VDS1022.app/Contents/Resources/lib/libusbJava.dylib` exists, so probably a permissions issue.
[image]

florentbr, 2022-03-04:
It looks like you installed the app from an emulated console for amd64.
The file "libusbJava.dylib" is 38Kb (amd64) when is should be 58Kb (arm64/aarch64).
Since your Java Runtime is `aarch64`, it can't load an `amd64` library.
Try to reinstall from a native console.

piannone, 2022-03-04:
Thank you sooooooo much @florentbr! That did it!
I forgot I had trouble with some packages and set iterm2 to run in rosetta.
I switched over to built-in terminal and it works!

closed 2022-03-04 by piannone