florentbr/OWON-VDS1022 #51: installing on opensue tumbleweed

Rickorich, 2022-04-04:
Hi
Does the installer support opensuse tumbleweed? I've tried to install and i get a message saying "install-linux.sh: line 140: dpkg-deb: command not found" which suggests that the installer has decided my system is deb based.
Thanks

florentbr, 2022-04-04:
The script selects the distribution based on the packet manager installed. In your case, it found `apt` which is supposed to be Debian and not openSuse. It should work once it selects `zypper` as `packager`.  It will require `rpm-build` to build a `rpm` package before install. Here's a modified install-linux.sh:

https://github.com/florentbr/OWON-VDS1022/blob/fix-opensuse/install-linux.sh
 
Let me know if it works.

Rickorich, 2022-04-05:
Hi. Thanks for the quick response. I changed  the installer script so whatever it thought my package manager was, it used zypper. It created the package and I installed it. Yast package manager complained about a missing dependency, libusb (I think it was), but it chose to continue regardless and the install worked. I was able to open the scope program and communicate with my scope. I am very impressed with the software. 

I would like to thank you for your work. I've worked in electronics for 30 odd years and never had my own scope!

I will try your installer as soon as I can.

florentbr, 2022-04-05:
> Yast package manager complained about a missing dependency, libusb (I think it was), but it chose to continue regardless and the install worked.

openSuse uses a different package name for libusb compared to other rpm packager. Changing `Requires: libusb1,` to `Requires: libusb-1.0.so.0,` should fix it:

https://github.com/florentbr/OWON-VDS1022/commit/089568638ec98145eee53584d98b45698720410d

> I would like to thank you for your work. I've worked in electronics for 30 odd years and never had my own scope!

Thanks.

florentbr, 2022-04-17:
fixed in release 1.1.3-cf16

closed 2022-04-17 by florentbr