florentbr/OWON-VDS1022 #52: New VDS1022s Don't Work

ajlenze, 2022-04-16:
I recently received 2 new VDS1022s from a vendor on AliExpress.  Their serial numbers are VDS10222207012 and VDS10222207005.  At first I couldn't get them to work with either Owon's Windows software or your alternate software, but after downloading and installing the latest versions of both (VDS 1.1.3 (build 20220314) for Owon's Windows software and 1.1.1-cf15 for your alternate software), I was able to get these new scopes working under Windows.  However, I was still not able to get them to work with your alternate software.  When I plug them in, it seems to go through the "Initializing..." process just fine, but no scope trace appears.  Also, if I  connect a CH1 scope probe to the 5V 1kHz test points then click on AutoSet, the "AutoSetting...Don't disconnect connection." message is displayed, but the autoset process never completes.

In case you don't think I have your software installed correctly, it works fine with an older VDS1022 that I bought on eBay.  (Its serial number is VDS10221827479.)  I'm running your software under Linux (Xubuntu), and it's worked great in the past on a mixture of VDS1022s, VDS1022Is, and Multcomp Pro MP720017s, which seems to be functionally equivalent to the VDS1022I.

In case you think the problem might be with my Linux (Xubuntu) installation, I also installed your software on the same Windows PC, which these new VDS1022s work fine with under the latest Owon software, and I had the exact same problem under the latest version of your software (the new VDS1022s don't work but an older VDS1022 does work).

Let me know if there's any other combination of software and scope that you want me to try, as I'm willing to do what I can to help you track down this problem.

florentbr, 2022-04-17:
What is the version number displayed in the About dialog ?
It's likely due to a new FPGA firmware added by OWON (see #38).

ajlenze, 2022-04-17:
V5.0.0

florentbr, 2022-04-17:
fixed in release 1.1.3-cf16

closed 2022-04-17 by florentbr

ajlenze, 2022-04-17:
I tested the new release with four different new VDS1022s (I received two
more yesterday) and one old one, and everything works perfectly (as long as
the idiot user - ME - doesn't connect the scope probe to the MULTI port).
Thanks for the quick response!

A.J.