florentbr/OWON-VDS1022 #56: Driver error looping on M1/Monterney

BorisDess, 2022-07-30:
```
java.lang.NoClassDefFoundError: Could not initialize class ch.ntb.usb.LibusbJava
	at com.owon.uppersoft.vds.core.usb.CDevice.scanMatchedDevices(CDevice.java:751)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:108)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:93)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:31)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:30)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:74)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:37)
```

florentbr, 2022-07-30:
This exception usually occur when the architecture is different between Java and the USB library.
Make sure that your JRE and the terminal used to install are both `aarch64` and not Rosetta emulated.
Have a look at #50

BorisDess, 2022-07-30:
Thanks a lot for the head's up ! 
Same problem, but reversed situation.
I had a x64 Java on an M1 system.
I got to replace the arm drivers with the x64 drivers, from the install folder to the ressources folder in the *.app.
And voilà !

closed 2022-07-30 by florentbr