florentbr/OWON-VDS1022 #57: Bode Plot

LeonIndman, 2022-08-04:
Hi!
Is it possible to implement Bode Plot for this device?

florentbr, 2022-08-04:
To plot the frequency response, the software would need to sweep the frequencies. 
How is it supposed to do it since this device doesn't have a signal generator?

florentbr, 2022-08-04:
The VDS1022 has the same fake Bode Plot capabilities as the one from your link.
It's called roll mode or slow mode. I wouldn't call it a Bode plot given that it doesn't have a log scale.
It's also limited to the current time-range and buffer size.
I'm not sure where you are going with this.

LeonIndman, 2022-08-04:
Thanks, you are probably right.

florentbr, 2022-08-04:
I made an error in my last comment, it's not slow/roll mode but low sweep mode.
Here's one interesting video from EEVBlog on the subject:
https://www.youtube.com/watch?v=uMH2hGvqhlE

LeonIndman, 2022-08-04:
Does the DSO5102 have a similar BodePlot mode
[image]
?

florentbr, 2022-08-04:
> Does the DSO5102 have a similar BodePlot mode?

I don't have one, but from your capture it just gives the Phase angle and the Gain for a single frequency.

florentbr, 2022-08-17:
I'm closing this request since it's too broad.

The application doesn't give the gain or phase shift but it can be easily measured with the cursors.

If your goal is to tune a cut-off frequency, then sweep the frequencies over 1 second with a sine generator or with a sound card. To visualize, setup the VDS1022 with a 50ms timebase, normal trigger and peak sampling.

If it's to plot a Bode graph, then open a Jupyter notebook and record/plot the gain and phase in a loop while manually swiping the frequencies.

closed 2022-08-17 by florentbr