florentbr/OWON-VDS1022 #58: Not detecting scope that does show up in lsusb

BillyCroan, 2022-09-08:
My owon oscilliscope is connected.  it shows up in lsusb, but the app says "OFFLINE" "No available usb found".
Fedora 36 x86_64, 
[image]

```
[root@hostname ~]# owon-vds-tiny
env: Linux, Java 1.8.0_281 amd64
app dir: /opt/owon-vds-tiny
user dir: /root/.owon-vds-tiny
locale: en_US, en_US, true
ch.ntb.usb.USBException: LibusbJava.usb_open: Input/output error
	at com.owon.uppersoft.vds.core.usb.CDevice.open(CDevice.java:85)
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:301)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
[Send @MACHINE_TYPE_ADD 0x56] recv -110 bytes
LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56] recv -110 bytes
LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56] recv -110 bytes
LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56] recv -110 bytes
LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56] recv -110 bytes
LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56] recv -110 bytes
LibusbJava.usb_bulk_read: Connection timed out
```
(repeats indefinitely)

I've tried running as my normal user and as root.
How do I enable ethernet mode?  I'd prefer that anyway to be honest.

Or how can I go about identifying why it's timing out on my scope?

I also captured usbmon0 in wireshark while attempting to connect.  (Attached)
[usb-oscilloscope-capture.zip](https://github.com/florentbr/OWON-VDS1022/files/9530169/usb-oscilloscope-capture.zip)

florentbr, 2022-09-08:
> How do I enable Ethernet mode? I'd prefer that anyway to be honest.

The VDS1022 doesn't have an Ethernet port.
What makes you think it's possible?

> Or how can I go about identifying why it's timing out on my scope?

What is your exact device version/serial ?
Do you get the same error with the Python API ?
Have you tried on Window or a different Linux distribution?

BillyCroan, 2022-09-10:
I have a VDS3104.  I guess I was just assuming all Owon oscilliscops of the vds series used the same viewer.  And the software GUI specifically shows a greyed out Ethernet option.  There's an RJ-45 port on my VDS3104 that is labeled LAN on the case.

My device serial is VDS31041418080.  on the barcoded sticker anyway

I'm not sure I am running the API correctly:
[bcroan@Beklarn python]$ pwd
/home/bcroan/gits/OWON-VDS1022/api/python
[bcroan@host python]$ python setup.py
/home/bcroan/gits/OWON-VDS1022/api/python/setup.py:13: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.core import setup
/usr/lib64/python3.10/distutils/dist.py:274: UserWarning: Unknown distribution option: 'include_package_data'
  warnings.warn(msg)
/usr/lib64/python3.10/distutils/dist.py:274: UserWarning: Unknown distribution option: 'python_requires'
  warnings.warn(msg)
/usr/lib64/python3.10/distutils/dist.py:274: UserWarning: Unknown distribution option: 'install_requires'
  warnings.warn(msg)
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied
[bcroan@host python]$ 

I haven't tried windows and don't have a windows system readily available.

florentbr, 2022-09-10:
It won't work with the VDS3104.
The software is different.
It's for the VDS1022 only.

closed 2022-09-10 by florentbr