florentbr/OWON-VDS1022 #59: 1.1.3-cf18 release quits on launch on OSX 10.11.6

nikbry, 2022-09-17:
Hi, have been running release 1.0.30-cf10 on OSX 10.11.6 with no issues. Updated to 1.1.3-cf18, which quits right after launch.
Looked at closed Issue re: Big Sur, but line 66 in install-mac.sh doesn't seem to be the culprit.
Cheers!

florentbr, 2022-09-17:
What is the output when the application is launched from a console ?
Can you post the detail of the installed files (`ls -lR "/Applications/OWON VDS1022.app"`) ?

nikbry, 2022-09-17:
Hi, I've attached two txt files with the console output and the installation files.

Regards.

> 
> 
> What is the output when the application is launched from a console ?
> Can you post the detail of the installed files (ls -lR "/Applications/OWON VDS1022.app") ?

Last login: Sat Sep 17 12:39:32 on ttys000
/Applications/OWON\ VDS1022.app/Contents/MacOS/launch ; exit;
Cally-2:~ Luis$ /Applications/OWON\ VDS1022.app/Contents/MacOS/launch ; exit;
env: Mac OS X, Java 1.8.0_31 x86_64
app dir: /Applications/OWON VDS1022.app/Contents/Resources
user dir: /Users/Luis/Library/Application Support/OWON VDS1022
locale: en_US, en_US, true
dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
  Referenced from: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _clock_gettime
  Referenced from: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

/Applications/OWON VDS1022.app/Contents/MacOS/launch: line 3:   546 Trace/BPT trap: 5       /usr/libexec/java_home --exec java -Xdock:name='OWON VDS1022' -Xdock:icon=icon.icns -cp 'lib/*' com.owon.vds.tiny.Main
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

Cally-2:Applications Luis$ ls -lR "OWON VDS1022.app"
total 0
drwxr-xr-x  5 root  admin  170 Sep 17 12:37 Contents

OWON VDS1022.app/Contents:
total 8
-rw-r--r--  1 root  admin  1488 Sep 17 12:37 Info.plist
drwxr-xr-x  3 root  admin   102 Sep 17 12:37 MacOS
drwxr-xr-x  8 root  admin   272 Sep 17 12:37 Resources

OWON VDS1022.app/Contents/MacOS:
total 8
-rwxr-xr-x  1 root  admin  186 Sep 17 12:37 launch

OWON VDS1022.app/Contents/Resources:
total 112
drwxr-xr-x  2 root  admin     68 Sep 17 12:37 api
drwxr-xr-x  7 root  admin    238 Sep 17 12:37 doc
drwxr-xr-x  8 root  admin    272 Sep 17 12:37 fwr
-rwxr-xr-x  1 root  admin  51232 Sep 17 12:37 icon.icns
drwxr-xr-x  9 root  admin    306 Sep 17 12:37 lib
-rwxr-xr-x  1 root  admin     10 Sep 17 12:37 version.txt

OWON VDS1022.app/Contents/Resources/api:

OWON VDS1022.app/Contents/Resources/doc:
total 6248
-rwxr-xr-x  1 root  admin  2501507 Sep 17 12:37 manual_en.pdf
-rwxr-xr-x  1 root  admin   694006 Sep 17 12:37 scpi_protocol_en.pdf
drwxr-xr-x  7 root  admin      238 Sep 17 12:37 tips_de
drwxr-xr-x  7 root  admin      238 Sep 17 12:37 tips_en
drwxr-xr-x  7 root  admin      238 Sep 17 12:37 tips_ru

OWON VDS1022.app/Contents/Resources/doc/tips_de:
total 40
-rwxr-xr-x   1 root  admin  1800 Sep 17 12:37 chl1.htm
drwxr-xr-x  13 root  admin   442 Sep 17 12:37 image
[... 57 lines cut ...]
-rwxr-xr-x  1 root  admin  23931 Sep 17 12:37 per3.png
-rwxr-xr-x  1 root  admin    658 Sep 17 12:37 selball.png
-rwxr-xr-x  1 root  admin  22683 Sep 17 12:37 trg1.png
-rwxr-xr-x  1 root  admin  24157 Sep 17 12:37 trg2.png
-rwxr-xr-x  1 root  admin  36482 Sep 17 12:37 trg3.png

OWON VDS1022.app/Contents/Resources/fwr:
total 2248
-rwxr-xr-x  1 root  admin  149554 Sep 17 12:37 VDS1022_FPGAV1_V3.9.bin
-rwxr-xr-x  1 root  admin  169216 Sep 17 12:37 VDS1022_FPGAV2_V4.1.bin
-rwxr-xr-x  1 root  admin  221368 Sep 17 12:37 VDS1022_FPGAV3_gaoyun.bin
-rwxr-xr-x  1 root  admin  221368 Sep 17 12:37 VDS1022_FPGAV4_gaoyun.bin
-rwxr-xr-x  1 root  admin  221368 Sep 17 12:37 VDS1022_FPGAV5_gaoyun.bin
-rwxr-xr-x  1 root  admin  149554 Sep 17 12:37 VDS2052_FPGA_v1.0.8.bin

OWON VDS1022.app/Contents/Resources/lib:
total 7488
-rwxr-xr-x  1 root  admin    58910 Sep 17 12:37 ch.ntb.usb-0.5.9.jar
-rwxr-xr-x  1 root  admin   259083 Sep 17 12:37 gson-2.7.0.jar
-rwxr-xr-x  1 root  admin   720702 Sep 17 12:37 jxl-2.6.6.jar
-rwxr-xr-x  1 root  admin    21200 Sep 17 12:37 libusb-0.1.4.dylib
-rwxr-xr-x  1 root  admin   106864 Sep 17 12:37 libusb-1.0.0.dylib
-rwxr-xr-x  1 root  admin    37548 Sep 17 12:37 libusbJava.dylib
-rwxr-xr-x  1 root  admin  2611754 Sep 17 12:37 owon-vds-tiny-1.1.3-cf18.jar
Cally-2:Applications Luis$

florentbr, 2022-09-17:
The issue is with the dependent `libusb-1.0.0.dylib` library. According to your logs, It fails with MacOS El Capitain due to a missing symbol at OS level.

https://stackoverflow.com/questions/45833569/dyld-lazy-symbol-binding-failed-symbol-not-found-clock-gettime-in-mongodb

If it used to work, then just replacing `libusb-1.0.0.dylib` with the one from release `1.0.30-cf10` should fix it.
Keep the latest of `libusb-0.1.4.dylib` and `libusbJava.dylib`.

You could also try to install [Homebrew libusb-compat](https://formulae.brew.sh/formula/libusb-compat).
The application tries to load `/usr/local/opt/libusb/lib/libusb-1.0.0.dylib` first and falls back to `OWON VDS1022.app/Contents/Resources/lib/libusb-1.0.0.dylib`.

nikbry, 2022-09-17:
Replacing the file worked like a charm!

Thanks a lot!

closed 2022-09-17 by nikbry