florentbr/OWON-VDS1022 #60: USB Connectivity MacOS12/m1

AlessandroEmm, 2022-10-15:
Hey there

First thanks a lot for your work!

I got my device very recently and i just can't seem to discover the device via USB. Do you know, whether it could be a permission issue? Google led me nowhere. The status LED switches between red and green, if that helps.

Thanks a lot!
Alessandro

florentbr, 2022-10-15:
Have a look at #50, #56, #59 .
It's probably a different architecture between Java and the installed library.
Make sure to install from a native aarch64/arm64 console (non Rosetta). Java needs to be aarch64/arm64 as well.
If it still doesn't work, post the logs.

AlessandroEmm, 2022-10-15:
Naturally you were right. I looked into docs/Readme but for some reason didn't think about looking at closed issues.  Maybe a line worthy for the docs?

Thanks!