florentbr/OWON-VDS1022 #63: streaming to pc memory

Sasanmaghami, 2022-12-29:
Hello dear florentbr.
thank you for your great work on owon's software.
Is it possible to tweak the softWare  so that it streams to pc memory so the sampling rate limitation be removed from higher time rames?
Thank you.

florentbr, 2022-12-29:
It should be possible to modify the software to accumulate/store the samples at lower sample rates.
It's quite the task since the software was not designed to support it and I'm not planning to spend free time on it.
Note that with the python API  you could accumulate continuous samples up to 100ks/s :

```python
from vds1022 import *

dev = VDS1022(debug=0)
dev.set_sampling('10k')  # 10K samples per second
dev.set_channel(CH1, range='10v', offset=2/10, probe='x10')

frames = dev.read('2s')  # accumulate samples during 2 second
frames.plot()
print("Samples: %s" % frames.ch1.size)
```

Sasanmaghami, 2022-12-29:
thank you mate.
i understand. truly you have already  given a lot of time and effort and honestly, done a lot more than owon for us.
even knowing that it is possible is hopeful.
i hope maybe some day there is a chance.
wish you best my friend.