florentbr/OWON-VDS1022 #64: Bug with FFT frequency for 100 MS/s sample rate

jasonRF90, 2023-01-13:
Hello.  I think I found a small bug with the frequency axis scaling of the FFT for timebases with 100MS/s.  I have attached a couple of examples where I am sending a 32 MHz sine wave into channel 1, as shown by the hardware frequency counter.  With 2 us/div timebase, the FFT peaks at 40 MHz, and with 500 ns/div timebase it peaks at 160 MHz.  To me it almost seems like the FFT axis is assuming the sample frequency is 5 kSamples / (20 * timebase), instead of taking into account the fact that at the 2 us/div  timebase there are only 4k samples and at the 500ns/div timebase there is 1k samples.  

Thanks,

Jason
[image]
[image]

florentbr, 2023-01-14:
Looks like a bug. Where did you get that there's only 4k samples at 2us/div and 1k samples at 500ns/div ?

jasonRF90, 2023-01-17:
> Looks like a bug. Where did you get that there's only 4k samples at 2us/div and 1k samples at 500ns/div ?

My primary reason for thinking that was because the *ADC? SCPI command only returns those shorter records, but I just looked at the documentation again and see that command only sends the samples that are on the display.    So I am probably wrong.  If you see 5k samples then I am definitely wrong!

Thanks.