florentbr/OWON-VDS1022 #65: Bug in mac version: Tips window cannot be permanently disabled

girishji, 2023-01-14:
When app is launched a "Tips" window appears. It has a "don't show it again" button but it does not work.

florentbr, 2023-01-14:
The "disable tips" is written in the settings, but pressing `Restore` restores the settings and therefore restores the tips.
One way around it would be to check "disable tips" and to click "Store" in "Utility" to persist the current settings.

girishji, 2023-01-15:
This does not work. When I tried to click the box the whole window
disappears.
Please stop that Tips window from popping up at startup.

florentbr, 2023-01-15:
The Tips window  is supposed to disappear when "disable tips" is checked. It's the normal behaviour.
Could be that the app failed to save the settings in `~/Library/Application Support/OWON VDS1022`.
Please post the logs from launch to click to exit.
To view the logs, run `/Applications/OWON VDS1022.app/Contents/MacOS/launch` from a console.

girishji, 2023-01-17:
1) It does not save the settings. I attached the json file at the end.
2) There are errors in the log (see below)
3) There is no menu option to get the 'tips' window back, which is like a user manual rather than tips. Can you add an option to main menu (top bar) to open this tips window when necessary? I am aware there is a F12 shortcut.

 /Applications/OWON\ VDS1022.app/Contents/MacOS/launch
env: Mac OS X, Java 17.0.5 aarch64
app dir: /Applications/OWON VDS1022.app/Contents/Resources
user dir: /Users/gp/Library/Application Support/OWON VDS1022
locale: en_US, en_US, true
ch.ntb.usb.USBException: LibusbJava.usb_get_busses: No such file or directory
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:236)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
ch.ntb.usb.USBException: LibusbJava.usb_get_busses: Unknown error: 316
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:236)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)

=====
VDS10222208010-cals.json:

{
  "cals": [
    [
      [
        1267,
        1102,
        972,
        804,
        675,
        547,
        979,
        849,
        722,
        550
      ],
      [
        1304,
        1107,
        980,
        811,
        683,
        553,
        984,
        857,
        731,
        556
      ]
    ],
    [
      [
        356,
        355,
        355,
        354,
        354,
        354,
        354,
        353,
        354,
        354
      ],
      [
        367,
        366,
        367,
        365,
        366,
        365,
        366,
        365,
        364,
        365
      ]
    ],
    [
      [
        1417,
        1432,
        1434,
        1436,
        1436,
        1436,
        1432,
        1434,
        1436,
        1436
      ],
      [
        1432,
        1437,
        1436,
        1436,
        1432,
        1433,
        1434,
        1435,
        1432,
        1433
      ]
    ]
  ]
}

[image]