florentbr/OWON-VDS1022 #68: remembering set-up with software re-start

carlos1w, 2023-01-24:
Is it possible to save the configuration that is running (i.e., which items are shown in the "measure") from one session to the next?  It seems that it always resets to showing only Vrms.  I would like to keep in the screen more information, but it is forgotten when the software restarts a different day.

Thank you for creating this wonderful software.

florentbr, 2023-01-25:
Click on "Store" in `Utility`. It will overwrite the default settings. Current settings are saved in the user folder when the application exists.

carlos1w, 2023-01-26:
> Click on "Store" in `Utility`. It will overwrite the default settings. Current settings are saved in the user folder when the application exists.

Thank you.

carlos1w, 2023-01-26:
> Click on "Store" in `Utility`. It will overwrite the default settings. Current settings are saved in the user folder when the application exists.

Thank you. 

The solution offered does not work, unfortunately.  No setting is saved, when I restart the Owon (mac, latest version) it comes back to only showing Vrms for both channels and nothing else.

carlos1w, 2023-01-31:
As an added comment: it seems the Windows version does save the configuration (tested on 1 computer) but the mac does not (tested on 3 macs).

carlos1w, 2023-01-31:
There is a folder "~/Library/Application\ Support/OWON\ VDS1022/" but it contains nothing, ever after the "Store" is pressed.  It is not a permissions issue, as I deleted this folder and it was re-created by the software, but with no content inside.

florentbr, 2023-01-31:
The settings are saved when the application exists. Your issue seems to be related to Mac since it works on Windows and Linux. Can you post the logs from start to exit? To view the logs run `/Applications/OWON VDS1022.app/Contents/MacOS/launch` in a console.

carlos1w, 2023-01-31:
Thank you, Florentbr.

Here is the output of the first run (at some point I activate CH2, add the various "measure" options, press the "store" button, and then move the baseline of CH1 and CH2).

[1st run.txt](https://github.com/florentbr/OWON-VDS1022/files/10550193/1st.run.txt)

Then I quit the application and restart.   and CH2 is inactive and the "measure" options are back to defaults.  Here is the output:

[2nd run.txt](https://github.com/florentbr/OWON-VDS1022/files/10550207/2nd.run.txt)