florentbr/OWON-VDS1022 #69: Averaging.

Duncan-dx1, 2023-02-03:
Please could you change averaging? Really if you average 4^n times you gain n bits of resolution. But the displayed resolution stays at 8 bits. It would be helpful to increase the resolution of the displayed waveform when averaging.
See VDS1022 teardown on EEV page 31 where I uploaded some trials.
Averaging could also optionally be used on normal display of waveform, for example by averaging 16/64 sequential samples. This is a low pass filter so loses fast data, but can improve visualisation of slow changes.
In both methods of averaging you could display changes or glitches above the noise level, for example to show parasitics, or to show random data as random data, not as the average of random data.
You could also apply both types of average. 
I was going to try and do it (issue 9). However I would likely be out of my comfort zone, never having used java.

[image]
[image]
[image]

Duncan-dx1, 2023-02-03:
An approach might be to display average and max/min in lighter colours. Then you could see parasitics and random data.

florentbr, 2023-02-03:
I should be possible but it would require a significant amount of work on the existing code. I don't have spare time for this feature, so you'll have to implement it yourself if you want it.

Duncan-dx1, 2023-02-06:
Thank you for you massive effort to date. I understand it is too much work.
I will try and compile and have a look at it.

Trace-2, 2023-02-06:
This is a job for ChatGPT LOL ... :)