florentbr/OWON-VDS1022 #72: Pulse width traceback

dizcza, 2023-02-13:
## Issue 1

I see many error logs when I select a pulse width trigger (> 400 us).
```
$ owon-vds-tiny 
env: Linux, Java 1.8.0_352 amd64
app dir: /opt/owon-vds-tiny
user dir: /home/dizcza/.owon-vds-tiny
locale: en, en_US, false
[Send @MACHINE_TYPE_ADD 0x56] recv V 0x1
found usb device VDS1022 idVendor:0x5345 idProduct:0x1234
[Send @FPGA_DOWNLOAD_QUERY_ADD 0x0] recv  0x1
fpga loaded: 1
read flash
[Send @read_flash 0x1] recv 2002 bytes
flash size: 2002
flash crc32: 1218904401
flash header: 0x55 0xAA
flash oem: true
flash board version: V5.0.1
flash board serial: VDS1022I2212909
load calibration /home/dizcza/.owon-vds-tiny/VDS1022I2212909-cals.json
Gain CH0 : [1269, 1105, 976, 809, 679, 549, 981, 854, 725, 554]
Gain CH1 : [1269, 1101, 972, 803, 676, 546, 976, 848, 721, 549]
Step CH0 : [358, 358, 358, 357, 357, 358, 358, 358, 358, 358]
Step CH1 : [357, 356, 356, 356, 356, 356, 355, 356, 355, 355]
Compensation CH0 : [1416, 1432, 1434, 1434, 1432, 1433, 1433, 1433, 1432, 1433]
Compensation CH1 : [1408, 1416, 1418, 1420, 1419, 1417, 1418, 1418, 1418, 1417]
[Send @trg_holdoff_ext_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch1_ADD 0x3842] recv S 0x0
[Send @TRG_ADD 0xffffc000] recv S 0x0
[Send @trg_holdoff_ch2_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch2_ADD 0xffffc7d1] recv S 0x0
[Send @channel_ch1_ADD 0xffffff80] recv S 0x0
[Send @volt_gain_ch1_ADD 0x2a7] recv S 0x0
[Send @zero_off_ch1_ADD 0x4e5] recv S 0x0
[Send @CHL_ON_ADD 0x3] recv S 0x0
[Send @channel_ch2_ADD 0xffffffa0] recv S 0x0
[Send @volt_gain_ch2_ADD 0x222] recv S 0x0
[Send @zero_off_ch2_ADD 0x63b] recv S 0x0
[Send @DM_ADD 0x13ec] recv S 0x0
[Send @SYNCOUTPUT_ADD 0x0] recv S 0x0
[Send @TIMEBASE_ADD 0x1] recv S 0x0
[Send @SAMPLE_ADD 0x0] recv S 0x0
[Send @SLOWMOVE_ADD 0x0] recv S 0x0
[Send @PRE_TRG_ADD 0x83e] recv S 0x0
[Send @SUF_TRG_ADD 0xbae] recv S 0x0
[Send @EMPTY_ADD 0x1] recv S 0x4e554c4c
[Send @pulse_level_ch1_ADD 0x525c] recv S 0x0
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x525c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x525c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @CHL_ON_ADD 0x1] recv S 0x4
[Send @channel_ch2_ADD 0x20] recv S 0x0
[Send @volt_gain_ch2_ADD 0x222] recv S 0x0
[Send @zero_off_ch2_ADD 0x63b] recv S 0x0
[Send @trg_holdoff_ext_ADD 0xffff8002] recv S 0x0
[Send @TRG_ADD 0xffffc000] recv S 0x0
[Send @trg_holdoff_ch2_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch2_ADD 0xffffc7d1] recv S 0x0
[Send @EMPTY_ADD 0x0] recv S 0x4e554c4c
[Send @pulse_level_ch1_ADD 0x525c] recv S 0x0
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x2a34] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @TRG_ADD 0x2000] recv S 0x4
[Send @trg_holdoff_ch2_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch2_ADD 0xfffff600] recv S 0x0
[Send @TRG_ADD 0x0] recv S 0x4
[Send @trg_holdoff_ext_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x0
[Send @pulse_level_ch1_ADD 0x535d] recv S 0x4
[Send @pulse_level_ch1_ADD 0x20c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x525c] recv S 0x4
[Send @pulse_level_ch1_ADD 0x2a34] recv S 0x4
java.lang.Exception: Tiny.Pulse.call: new
	at com.owon.uppersoft.vds.source.comm.ext.ClockTimeAdjuster.c_trg_condtionArg(ClockTimeAdjuster.java:112)
	at com.owon.uppersoft.vds.source.comm.Submitor2.c_trg_pulse(Submitor2.java:459)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.submit_trg_addition(TinyTrgSubmitHandler.java:207)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.handleSubmit_trg_type(TinyTrgSubmitHandler.java:97)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.selfSubmit_Tiny(TinyTrgSubmitHandler.java:59)
	at com.owon.uppersoft.dso.global.CoreControlTiny$1.selfSubmit(CoreControlTiny.java:65)
	at com.owon.uppersoft.dso.model.trigger.TriggerControl.doSubmit(TriggerControl.java:498)
	at com.owon.uppersoft.dso.model.trigger.TriggerUIInfo.c_setTrigger(TriggerUIInfo.java:84)
	at com.owon.uppersoft.dso.view.trigger.TrgModeItemListener.handleSelect(TrgModeItemListener.java:45)
	at com.owon.uppersoft.vds.source.comm.TrgModeItemListener_Tiny.handleSelect(TrgModeItemListener_Tiny.java:27)
	at com.owon.uppersoft.dso.view.trigger.TrgModeItemListener.itemStateChanged(TrgModeItemListener.java:34)
	at javax.swing.JComboBox.fireItemStateChanged(JComboBox.java:1225)
	at javax.swing.JComboBox.selectedItemChanged(JComboBox.java:1282)
	at javax.swing.JComboBox.contentsChanged(JComboBox.java:1328)
	at javax.swing.AbstractListModel.fireContentsChanged(AbstractListModel.java:118)
	at javax.swing.DefaultComboBoxModel.setSelectedItem(DefaultComboBoxModel.java:93)
	at javax.swing.JComboBox.setSelectedItem(JComboBox.java:578)
	at javax.swing.JComboBox.setSelectedIndex(JComboBox.java:624)
	at javax.swing.plaf.basic.BasicComboPopup$Handler.mouseReleased(BasicComboPopup.java:868)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:290)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at javax.swing.plaf.basic.BasicComboPopup$1.processMouseEvent(BasicComboPopup.java:517)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at org.GNOME.Accessibility.AtkWrapper$6.dispatchEvent(AtkWrapper.java:717)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
[Send @TRG_ADD 0x4100] recv S 0x4
[Send @trg_holdoff_ext_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch1_ADD 0xf19] recv S 0x0
[Send @trg_cdt_gl_ch1_ADD 0x0] recv S 0x0
[Send @trg_cdt_hl_ch1_ADD 0x0] recv S 0x0
```

## Issue 2
The app becomes unresponsive when I drag the green left cursor up and down while in the sampling mode. Only restarting helps.

florentbr, 2023-02-13:
I don't have this issue on Linux/KDE desktop. It looks like it's related to the Gnome desktop and Java. Can you try with Java 11 or 17 ?

dizcza, 2023-02-13:
With Java 11:
```
$ owon-vds-tiny 
env: Linux, Java 11.0.17 amd64
app dir: /opt/owon-vds-tiny
user dir: /home/dizcza/.owon-vds-tiny
locale: en, en_US, false
Gtk-Message: 18:52:03.896: Failed to load module "canberra-gtk-module"
[Send @MACHINE_TYPE_ADD 0x56] recv V 0x1
found usb device VDS1022 idVendor:0x5345 idProduct:0x1234
[Send @FPGA_DOWNLOAD_QUERY_ADD 0x0] recv  0x1
fpga loaded: 1
read flash
[Send @read_flash 0x1] recv 2002 bytes
flash size: 2002
flash crc32: 1218904401
flash header: 0x55 0xAA
flash oem: true
flash board version: V5.0.1
flash board serial: VDS1022I2212909
load calibration /home/dizcza/.owon-vds-tiny/VDS1022I2212909-cals.json
Gain CH0 : [1269, 1105, 976, 809, 679, 549, 981, 854, 725, 554]
Gain CH1 : [1269, 1101, 972, 803, 676, 546, 976, 848, 721, 549]
Step CH0 : [358, 358, 358, 357, 357, 358, 358, 358, 358, 358]
Step CH1 : [357, 356, 356, 356, 356, 356, 355, 356, 355, 355]
Compensation CH0 : [1416, 1432, 1434, 1434, 1432, 1433, 1433, 1433, 1432, 1433]
Compensation CH1 : [1408, 1416, 1418, 1420, 1419, 1417, 1418, 1418, 1418, 1417]
java.lang.Exception: Tiny.Pulse.call: new
	at com.owon.uppersoft.vds.source.comm.ext.ClockTimeAdjuster.c_trg_condtionArg(ClockTimeAdjuster.java:112)
	at com.owon.uppersoft.vds.source.comm.Submitor2.c_trg_pulse(Submitor2.java:459)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.submit_trg_addition(TinyTrgSubmitHandler.java:207)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.handleSubmit_trg_type(TinyTrgSubmitHandler.java:97)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.selfSubmit_Tiny(TinyTrgSubmitHandler.java:59)
	at com.owon.uppersoft.dso.global.CoreControlTiny$1.selfSubmit(CoreControlTiny.java:65)
	at com.owon.uppersoft.dso.global.CoreControl.initDetail(CoreControl.java:219)
	at com.owon.uppersoft.dso.global.ControlManager.initDetail(ControlManager.java:402)
	at com.owon.uppersoft.vds.source.comm.InterCommTiny.initDetail(InterCommTiny.java:258)
	at com.owon.uppersoft.vds.source.comm.InterCommTiny.initMachine(InterCommTiny.java:422)
	at com.owon.uppersoft.dso.source.comm.detect.InitLinkRunner.initLink(InitLinkRunner.java:72)
	at com.owon.uppersoft.dso.source.comm.detect.InitLinkRunner.run(InitLinkRunner.java:35)
	at com.owon.uppersoft.dso.source.comm.Flow.doStuff(Flow.java:48)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:66)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
[Send @TRG_ADD 0x4940] recv S 0x0
[Send @trg_holdoff_ext_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch1_ADD 0xf19] recv S 0x0
[Send @trg_cdt_gl_ch1_ADD 0xffff9bdc] recv S 0x0
[Send @trg_cdt_hl_ch1_ADD 0x0] recv S 0x0
[Send @channel_ch1_ADD 0xffffff80] recv S 0x0
[Send @volt_gain_ch1_ADD 0x2a7] recv S 0x0
[Send @zero_off_ch1_ADD 0x4e5] recv S 0x0
[Send @CHL_ON_ADD 0x1] recv S 0x0
[Send @channel_ch2_ADD 0x20] recv S 0x0
[Send @volt_gain_ch2_ADD 0x222] recv S 0x0
[Send @zero_off_ch2_ADD 0x63b] recv S 0x0
[Send @DM_ADD 0x13ec] recv S 0x0
[Send @SYNCOUTPUT_ADD 0x0] recv S 0x0
[Send @TIMEBASE_ADD 0x50] recv S 0x0
[Send @SAMPLE_ADD 0x0] recv S 0x0
[Send @SLOWMOVE_ADD 0x0] recv S 0x0
[Send @PRE_TRG_ADD 0x9f1] recv S 0x0
[Send @SUF_TRG_ADD 0x9fb] recv S 0x0
[Send @EMPTY_ADD 0x1] recv S 0x4e554c4c
```

florentbr, 2023-02-13:
What's your Linux distribution / version / desktop ?

```
Gtk-Message: 18:52:03.896: Failed to load module "canberra-gtk-module"
```

This could be the cause of your issue. Have a look at:
https://fr.mathworks.com/matlabcentral/answers/472134-gtk-message-10-32-31-466-failed-to-load-module-canberra-gtk-module

dizcza, 2023-02-14:
Ubuntu 22.04 GNOME.

> Gtk-Message: 18:52:03.896: Failed to load module "canberra-gtk-module"

This is a warning and I often see this message while trying to run other apps. It's not an error and you can ignore it. Anyhow, I followed your link, installed everything and the warning disappeared. The traceback not gone.

dizcza, 2023-02-14:
My Java 11 is OpenJDK, not Oracle's. Don't know if this could be the reason.

florentbr, 2023-02-14:
I run OpenJDK as well and I don't have this issue.

I had a quick look at the source and the exception in your logs is just a statement meant to print the stack trace for debugging. It's a only called with the newer devices (version > 2.5) and can be ignored. I'll remove it in the next release.

Regarding issue 2, can you indicate at which line it occurs in your logs ?
A step by step use case would also help.

Maby printing the stack trace (issue 1) is taking too much resources and crippling the execution, but I doubt it. I'll try to run some tests this weekend.

dizcza, 2023-02-15:
All right, after I switched to Java 11 I no longer experience issue 2. I do however see some tracebacks but they don't seem to harsh the app. Here is one of them I got when changing the time axis div/s.

```
[Send @TIMEBASE_ADD 0x13880] recv S 0x4
[Send @SAMPLE_ADD 0x0] recv S 0x0
[Send @SLOWMOVE_ADD 0x1] recv S 0x0
[Send @PRE_TRG_ADD 0x9f6] recv S 0x0
[Send @SUF_TRG_ADD 0x9f6] recv S 0x0
[Send @EMPTY_ADD 0x1] recv S 0x4e554c4c
[Send @TIMEBASE_ADD 0x9c40] recv S 0x4
[Send @SAMPLE_ADD 0x0] recv S 0x0
[Send @SLOWMOVE_ADD 0x1] recv S 0x0
[Send @PRE_TRG_ADD 0x9f6] recv S 0x0
[Send @SUF_TRG_ADD 0x9f6] recv S 0x0
[Send @EMPTY_ADD 0x1] recv S 0x4e554c4c
java.lang.Exception: Tiny.Pulse.call: new
	at com.owon.uppersoft.vds.source.comm.ext.ClockTimeAdjuster.c_trg_condtionArg(ClockTimeAdjuster.java:112)
	at com.owon.uppersoft.vds.source.comm.Submitor2.c_trg_pulse(Submitor2.java:459)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.submit_trg_addition(TinyTrgSubmitHandler.java:207)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.handleSubmit_trg_type(TinyTrgSubmitHandler.java:97)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.selfSubmit_Tiny(TinyTrgSubmitHandler.java:59)
	at com.owon.uppersoft.dso.global.CoreControlTiny$1.selfSubmit(CoreControlTiny.java:65)
	at com.owon.uppersoft.dso.model.trigger.TriggerControl.doSumbitTrgSweep(TriggerControl.java:504)
	at com.owon.uppersoft.dso.model.trigger.VoltsensableTrigger.c_setSweep(VoltsensableTrigger.java:163)
	at com.owon.uppersoft.dso.delegate.DefaultTimeControlDelegate.onTimebaseChange(DefaultTimeControlDelegate.java:102)
	at com.owon.uppersoft.dso.mode.control.TimeControl.internalTimebaseIndex(TimeControl.java:152)
	at com.owon.uppersoft.dso.mode.control.TimeControl.c_setTimebaseIdx(TimeControl.java:221)
	at com.owon.uppersoft.dso.view.sub.DetailPane.changeTimeBase(DetailPane.java:208)
	at com.owon.uppersoft.dso.view.sub.DetailPane.nextTimeBase(DetailPane.java:190)
	at com.owon.uppersoft.dso.control.ChartScreenMouseGesture.mouseWheelMoved(ChartScreenMouseGesture.java:328)
	at com.owon.uppersoft.dso.control.IntermediateMouseAdapter.mouseWheelMoved(IntermediateMouseAdapter.java:66)
	at java.desktop/java.awt.Component.processMouseWheelEvent(Component.java:6723)
	at java.desktop/java.awt.Component.processEvent(Component.java:6407)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4576)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
java.lang.Exception: Tiny.Pulse.call: new
	at com.owon.uppersoft.vds.source.comm.ext.ClockTimeAdjuster.c_trg_condtionArg(ClockTimeAdjuster.java:112)
	at com.owon.uppersoft.vds.source.comm.Submitor2.c_trg_pulse(Submitor2.java:459)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.submit_trg_addition(TinyTrgSubmitHandler.java:207)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.handleSubmit_trg_type(TinyTrgSubmitHandler.java:97)
	at com.owon.uppersoft.vds.source.comm.ext.TinyTrgSubmitHandler.selfSubmit_Tiny(TinyTrgSubmitHandler.java:59)
	at com.owon.uppersoft.dso.global.CoreControlTiny$1.selfSubmit(CoreControlTiny.java:65)
	at com.owon.uppersoft.dso.model.trigger.TriggerControl.doSubmit(TriggerControl.java:498)
	at com.owon.uppersoft.dso.delegate.DefaultTimeControlDelegate.onTimebaseChange(DefaultTimeControlDelegate.java:106)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
[Send @TRG_ADD 0x4140] recv S 0x4
[Send @trg_holdoff_ext_ADD 0xffff8002] recv S 0x0
[Send @pulse_level_ch1_ADD 0x202a] recv S 0x0
[Send @trg_cdt_gl_ch1_ADD 0xffffc2ec] recv S 0x0
[Send @trg_cdt_hl_ch1_ADD 0x0] recv S 0x0
```

florentbr, 2023-02-15:
These tracebacks (`java.lang.Exception: Tiny.Pulse.call: new`) are just log information. They are not the result of an issue or error and can be ignored.

dizcza, 2023-02-15:
> These tracebacks (`java.lang.Exception: Tiny.Pulse.call: new`) are just log information. They are not the result of an issue or error and can be ignored.

Well, they look like valid error messages. It would be nice to at least indicate that it's a log.

Anyhow, feel free to close the issue. Issue 2 was more annoying but it seems working now.

Oh, I forgot to ask. How do I set the MULTI EXT channel as the digital input trigger for CH1? Originally asked in https://www.eevblog.com/forum/testgear/owon-vds1022-ext-trigger/msg4699073.

florentbr, 2023-02-15:
> Well, they look like valid error messages. It would be nice to at least indicate that it's a log.

They are not supposed to be there. It's just a leftover from development/debugging to track the calls:
```java
(new Exception("Tiny.Pulse.call: new")).printStackTrace();
```

> Oh, I forgot to ask. How do I set the MULTI EXT channel as the digital input trigger for CH1? Originally asked in 

Utility -> Channel -> MULTI -> Select Trigger In

closed 2023-03-11 by dizcza