florentbr/OWON-VDS1022 #76: Raspberry Pi No available USB found Resource temporarily unavailable

aaronnewcomb, 2023-03-07:
I tried installing on a fresh install of Raspbian today. Unfortunately, it looks like maybe libusbjava isn't working somehow? I am getting "No available USB found" in the GUI and Java errors on the command line.
`env: Linux, Java 17.0.6 arm
app dir: /opt/owon-vds-tiny
user dir: /root/.owon-vds-tiny
locale: en, en_US, false
ch.ntb.usb.USBException: LibusbJava.usb_open: Resource temporarily unavailable (-1568584872)
	at com.owon.uppersoft.vds.core.usb.CDevice.open(CDevice.java:86)
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:300)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
`

aaronnewcomb, 2023-03-07:
I also tried installing the version based on Debian Buster. Same result.

florentbr, 2023-03-07:
Could you provide :
* the exact system architecture : `uname -a`
* the usb descriptor : `lsusb -v -s 5345`

aaronnewcomb, 2023-03-07:
[image]
lsusb -v -s 5345 does not return anything. However, here is what shows up about the device when I just use lsusb -v ...
[image]

aaronnewcomb, 2023-03-07:
BTW - I just tried this on a fresh install using a model 2B and it worked. Slowly of course, but it did work.

florentbr, 2023-03-07:
> lsusb -v -s 5345 does not return anything.

My bad, it should be `lsusb -v -d 5345:1234`

> I tried installing on a fresh install of Raspbian today. Unfortunately, it looks like maybe libusbjava isn't working somehow?

The native `libusbjava.so`  provided here was compiled for `armhf`. It's likely the source of your issue since your system is `armv7l`. It should work once compiled to the correct architecture. Sources: https://github.com/florentbr/OWON-VDS1022/blob/master/lib/libusbJava-src.zip

> BTW - I just tried this on a fresh install using a model 2B and it worked. Slowly of course, but it did work.

Disable all the measures, it should improve the refresh rate.

aaronnewcomb, 2023-03-08:
I don't think that armv71 could be the issue since it is working on my 2B and that is also running armv71.
[image]

florentbr, 2023-03-08:
Have a look at the system events. Maybe a driver is crashing or interfering with libusb.
Note that one has a linux kernel  `v7+` while the other is `v7l+`. So it could still be related to the compiled architecture.

aaronnewcomb, 2023-03-08:
So, thinking this may be related to the architecture I decided to try 64bit Raspbian since that will be different again. And ... it works when compiled on this architecture. 
[image]
So, I am going to call this good enough for now. Perhaps this will resolve itself with future changes in the supporting architecture libraries. Maybe this will help someone else who is having a similar problem.

nistvan86, 2023-06-30:
@aaronnewcomb is this a Raspberry Pi 4B? Does the software work with usable speed on that? The 2B was too slow as you mentioned.
(Sorry for hijacking the thread, but there's not much information available on the net about this.)

aaronnewcomb, 2023-06-30:
Yes. it was a Raspberry Pi 4B. However, that was overkill. I ended up using an Orange Pi 3 board running their Ubuntu image and it works perfectly. I hooked it up to a 10" touchscreen and 3D printed a case to put everything in. It works like a champ and now I don't have to connect it to my desktop or laptop.

nistvan86, 2023-06-30:
@aaronnewcomb thank you! Tested the install on a Pi4 with recent 64bit RaspberryPi OS and it was straightforward. Ordered a scope, I'm looking forward to it. :)