florentbr/OWON-VDS1022 #83: Could not find the main class: lib\gson-2.7.jar on windows 10

mletombe, 2023-06-30:
Hello, 

First thank you for your work. 

It's working well on Ubuntu, but not on windows 10, I get the message in the title. 
I had an old java 6 installed, the program gave me the error in the title. I then installed a new jdk 17, same error, even after reboot. I didn't deinstall the java 6.

The official Owon VDS_C2 is working well on the same machine.

Mathieu.

florentbr, 2023-06-30:
Follow the instructions at [home page](https://github.com/florentbr/OWON-VDS1022#windows-10-8-7-xp-x86amd64) and post the result.
Why install a JDK instead of a simple JRE ?
I'd try to clean the `PATH` and `JAVA_HOME` environment variables and then install a JRE and the APP.

mletombe, 2023-06-30:
It works!

I did deinstall the software and JDK 17, clean JAVA_HOME (it was the old 6), install JRE 17, reboot, reinstall your software, and it works !
I did install JDK because this the first thing coming when I clicked on your link toward adoptium, and I'm not great abour JAVA. 

Thanks a lot!

mletombe, 2023-07-05:
ok