florentbr/OWON-VDS1022 #84: Add 100v/div for the isolated version when using a x10 attenuation probe

fakuivan, 2023-08-26:
The isolated version supports up to 400V peak to peak meaning that with a x10 attenuation probe it should be possible and somewhat safe to measure AC mains.

ceres-c, 2023-08-26:
Pretty sure the proves that come with the thing are no way safe at mains voltage, so I’m not sure this might be a good  idea

fakuivan, 2023-08-26:
The probes that come with the isolated version are specd up to 600V, if you want to believe that or not is a different thing.

florentbr, 2023-08-27:
The 400v rating is the peak to peak voltage it can withstand on the inputs without probe before it breaks the isolation barrier or damage components. It's not the measurable voltage. The input range it can measure is specified at 50v peak to peak without probe (vertical sensitivity in specification). You shouldn't use a 10x probe to measure main voltage directly. AC main in Argentina is 230v RMS which is 325v peak and  650v peak to peak. It's over the measurable range of this device after attenuation with a 10x probe.

fakuivan, 2023-08-27:
That's interesting, then that 400v limit must be specified assuming the ground potential is the same between the isolation barrier. Thanks for clarifying the input range thing.

florentbr, 2023-08-27:
> That's interesting, then that 400v limit must be specified assuming the ground potential is the same between the isolation barrier. Thanks for clarifying the input range thing.

The ground between probe and USB is different with the isolated version. At least it's the case with mine. I have no continuity between probe ground and usb/casing ground.

fakuivan, 2023-08-27:
Of course that's because it's isolated, but the if the isolation is rated at 400V then that'd be assuming 0V at the USB gnd and 400V at the input, aka a max 400V between the USB and the analog lines, which is where the isolated dc-dc converter and the signal isolator sits between. I'm not saying that both grounds are coupled, that's the point of the isolation :P, but they they might have been coupled at the time of testing the isolation barrier and rating it at 400V.

This was just an observation though, I'll close this since the 25V max amplitude has been clarified.

closed 2023-08-27 by fakuivan