florentbr/OWON-VDS1022 #85: M1 Macbook Air throws java.lang.UnsatisfiedLinkError

girishji, 2023-09-17:
Details below.

It used to work. Maybe new Macos updates broke it. 

libusbJava.dylib is present.

```
...WON VDS1022.app/Contents/Resources/lib % ll
total 7712
-rw-r--r--@ 1 root  admin    58910 Sep 17 16:11 ch.ntb.usb-0.5.9.jar
-rw-r--r--@ 1 root  admin   259083 Sep 17 16:11 gson-2.7.0.jar
-rw-r--r--@ 1 root  admin   720702 Sep 17 16:11 jxl-2.6.6.jar
-rw-r--r--@ 1 root  admin    55920 Sep 17 16:11 libusb-0.1.4.dylib
-rw-r--r--@ 1 root  admin   159248 Sep 17 16:11 libusb-1.0.0.dylib
-rwxr-xr-x@ 1 root  admin    58557 Sep 17 16:11 libusbJava.dylib*
-rw-r--r--@ 1 root  admin  2622148 Sep 17 16:11 owon-vds-tiny-1.1.5-cf19.jar
```

When I start`/Applications/OWON\ VDS1022.app/Contents/MacOS/launch` I get this:

```
% /Applications/OWON\ VDS1022.app/Contents/MacOS/launch
env: Mac OS X, Java 19.0.2 x86_64
app dir: /Applications/OWON VDS1022.app/Contents/Resources
user dir: /Users/gp/Library/Application Support/OWON VDS1022
locale: en_US, en_US, true
java.lang.UnsatisfiedLinkError: Can't load library: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusbJava.dylib
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2408)
	at java.base/java.lang.Runtime.load0(Runtime.java:785)
	at java.base/java.lang.System.load(System.java:2011)
	at ch.ntb.usb.LibusbJava.load(LibusbJava.java:407)
	at ch.ntb.usb.LibusbJava.<clinit>(LibusbJava.java:386)
	at com.owon.uppersoft.vds.core.usb.CDevice.init(CDevice.java:213)
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:229)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
java.lang.NoClassDefFoundError: Could not initialize class ch.ntb.usb.LibusbJava
	at com.owon.uppersoft.vds.core.usb.CDevice.init(CDevice.java:213)
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:229)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.UnsatisfiedLinkError: Can't load library: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusbJava.dylib [in thread "Thread-1"]
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2408)
	at java.base/java.lang.Runtime.load0(Runtime.java:785)
	at java.base/java.lang.System.load(System.java:2011)
	at ch.ntb.usb.LibusbJava.load(LibusbJava.java:407)
	at ch.ntb.usb.LibusbJava.<clinit>(LibusbJava.java:386)
	... 8 more
java.lang.NoClassDefFoundError: Could not initialize class ch.ntb.usb.LibusbJava
	at com.owon.uppersoft.vds.core.usb.CDevice.init(CDevice.java:213)
	at com.owon.uppersoft.vds.core.usb.CDevice.getDevices(CDevice.java:229)
	at com.owon.uppersoft.dso.source.usb.USBSourceManager.refreshUSBPort(USBSourceManager.java:79)
	at com.owon.uppersoft.dso.source.comm.detect.USBLoopChecker.checkUSBDevice(USBLoopChecker.java:74)
	at com.owon.uppersoft.dso.source.comm.USBDaemonHelper.onNotConnecting(USBDaemonHelper.java:35)
	at com.owon.uppersoft.vds.machine.InfiniteDaemonTiny0.onNotConnecting(InfiniteDaemonTiny0.java:37)
	at com.owon.uppersoft.dso.source.comm.Flow.run(Flow.java:79)
	at com.owon.uppersoft.dso.global.ControlAppsTiny$1.run(ControlAppsTiny.java:38)
```

florentbr, 2023-09-18:
The architecture of the Macbook M1 is `aarch64`.  
Install  Java `aarch64` and not `x86_64`.

girishji, 2023-09-19:
Thank you, that was it.
I didn't know I already had a aarch64 JDK before I installed the Amazon's x86 version which caused the problem..

closed 2023-09-19 by girishji