florentbr/OWON-VDS1022 #86: install on Slackware

scboulogne, 2023-10-08:
1.1.5-cf19 installs and runs nicely on Slackware 14.2 (and likely 15) Linux x86_64
with the modified installation script, see the attached patch  (where I also corrected a typo there)
[install-linux.sh-patch.txt](https://github.com/florentbr/OWON-VDS1022/files/12840494/install-linux.sh-patch.txt) 
the tested scope model is MP720017 (reporting to lsusb as 5345:1234 Owon PDS6062T)

small remarks (unrelated to the main topic of this issue): 
1. my wm doesn't have window kill button, would be nice to have the way to quit the application nicely with a key binding? or a button?
2. one may note someplace in the README that running these gadgets on a Pi3 (and probably Pi4) requires additional power supplied through an usb Y-cable or a powered hub

florentbr, 2023-10-09:
> 1.1.5-cf19 installs and runs nicely on Slackware 14.2 (and likely 15) Linux x86_64
with the modified installation script, see the attached patch (where I also corrected a typo there)
install-linux.sh-patch.txt

The package for slackware is quite simple. There's no need to use a rpm converter especially since the dependencies are different. This should work (untested):

```bash
	local packager
	for packager in installpkg pacman dnf yum zypper ppm apt / ; do
		[ $packager = / ] && raise 'Package manager not supported'
		command -v $packager >/dev/null && break
	done
```

```bash
	case $packager in
		...
		installpkg) build-slackware $arch ;;
```

```bash
	case "$packager" in
		...
		installpkg) installpkg "$_PACKAGE" ;;
```

```bash
build-slackware () {
	
	local arch=$1

	write_files "$arch"

	case "$arch" in
		amd64) arch=x86_64  ;;
		arm64) arch=aarch64 ;;
	esac

	echo 'Build Slackware package ...'

	write install/slack-desc <<EOF
${_ID}: ${_ID} (${_NAME})
${_ID}: 
${_ID}: ${_SUMMARY}
${_ID}: 
${_ID}: ${_HOMEPAGE}
${_ID}: 
${_ID}: 
${_ID}: 
${_ID}: 
${_ID}: 
${_ID}: 
EOF

	write install/slack-required <<< "openjdk\nlibusb"
	write install/doinst.sh +x   <<< "${_POSTINSTALL}"
	write install/douninst.sh +x <<< "${_POSTREMOVE}"

	_PACKAGE="$_TMPDIR/$_ID-${_VERSION/-*/}-$arch-${_VERSION/*-/}.tgz"

	rm -f "$_PACKAGE"
	tar -czvf "$_PACKAGE" * >/dev/null || exit 1
}
```

> my wm doesn't have window kill button, would be nice to have the way to quit the application nicely with a key binding? or a button?

Your WM should have a shortcut to close any application. Have you tried Alt+F4 ?

> one may note someplace in the README that running these gadgets on a Pi3 (and probably Pi4) requires additional power supplied through an usb Y-cable or a powered hub

It's mentioned in the documentation of the device if I remember correctly.

scboulogne, 2023-10-10:
Ok, thanks. This should work fine and is more consistent with your install script. (Since it worked for me via rpm-tools, I didn't bother with this solution.)

Thank you for your rapid fix.

About the Y-cable: I am testing an MP720019 shipped with it by multicomp-pro (so not owon). They say nothing about power supply in their little booklet and I didn't check their cd assuming it was Windows software. Probably I should've, they might have the manual there. The unit worked fine on my slackware laptop, and froze the Pi3... It worked ok after I inserted a powered usb hub.

Btw, in this context, would you know how safe MP720019 is compared to a usual scope? And with regard to the computer it is connected to? I am considering to deploy them in a student class (3rd and 2nd year university physics) equipped with Pi3's. I understand about the 2 channels requiring a common ground. That is common practice anyway. However, what would happen if a student connects his/her ground probe stupidly to something else? The way I see it, it will propagate to the computer through the usb ground, which won't do any good to a little Pi3. Would an independently powered usb hub isolate the two?

scboulogne, 2023-10-11:
Thank you very much for your comments. 

Unfortunately, unlike laptops, pi's must use external displays. In my case, those are old VGA flat screens with integrated AC power supply (coming from the university computer class surplus). I didn't think of checking their VGA signal ground and the VGA->HDMI adapter ground. Will do that next time I go there. Another source of ground looping may come through the Ethernet cables. I suppose, that none of it materialized when I was testing the unit.

I will probably advise them of buying the MP720017 units for the lab.
Overall, I am quite satisfied with it and its integration in the pi classroom.
And yes, they do normally use current limiting power supplies or signal generators, but I have seen students doing very strange things, and tired teachers not having reacted to such mishaps timely. 

As for MP720019, they do have one available for testing (from Farnell). I didn't touch it because MP720017 suffices (and I have other things to do). It is 100MHz and twice the price. Any chance it gets supported one day?

scboulogne, 2023-10-17:
Ok, thanks. Closing this

closed 2023-10-17 by scboulogne