florentbr/OWON-VDS1022 #87: MacOS Sierra 10.12.6 x64 - Won't recognize USB Device OWON VDS 1022

Petertechmail, 2023-11-25:
I finally got the Owon app to launch in Sierra 10.12.6 x64 but the VDS Owon is not recognized as a usb device. Upper left hand screen says: OFFLINE. Any suggestions?

florentbr, 2023-11-25:
Can you provide the logs from the console? Is the device listed by the OS?
This issue is usually related to a different architecture between this app and the installed Java.
Make sure both of them are x64.

Petertechmail, 2023-11-25:
I have no idea how to copy the correct log file. Some guidance appreciated. I know how to open the console but not which part to copy. 

I followed the instructions on github for MacOS , listed as the latest version . I didn’t see different architecture versions unless there is a previous version available somewhere else.

florentbr, 2023-11-25:
Just post what's printed once the debug command line is executed. See home page for the command line. Executing `java -version` will print the Java version, and `lsusb` will list all the USB devices.

Petertechmail, 2023-11-28:
MacBookPro13s-MacBook-Pro:~ peter$ java -version
openjdk version "1.8.0_392"
OpenJDK Runtime Environment (Temurin)(build 1.8.0_392-b08)
OpenJDK 64-Bit Server VM (Temurin)(build 25.392-b08, mixed mode)
MacBookPro13s-MacBook-Pro:~ peter$

Petertechmail, 2023-11-28:
No output with lsusb:

MacBookPro13s-MacBook-Pro:~ peter$ lsusb
-bash: lsusb: command not found

Petertechmail, 2023-11-28:
Not sure which log:

[image]

Petertechmail, 2023-11-28:
[image]

Petertechmail, 2023-11-28:
Tried this:

MacBookPro13s-MacBook-Pro:~ peter$ /Applications/OWON\ VDS1022.app/Contents/MacOS/launch ; exit;
env: Mac OS X, Java 1.8.0_392 x86_64
app dir: /Applications/OWON VDS1022.app/Contents/Resources
user dir: /Users/peter/Library/Application Support/OWON VDS1022
locale: en_US, en_US, true
logout

florentbr, 2023-11-28:
There's no error so it's likely an hardware or OS issue. Could be that your device is drawing too much current.  Check the events from your OS and try with an external power supply. Also make sure that your OS is listing the device.

Petertechmail, 2023-11-30:
Thanks, I’ll look into it. I don’t see and external power connection unless it’s that 5V 1khz socketed. Otherwise I’ll have to use one of those usb cables with an extra usb power connector

wandering3ngineer, 2023-12-30:
Similar problem wtih Ubuntu Linux 20.04. GUI interface shows **OFFLINE** in upper left corner. **No available usb found** ... I'm connecting HDS2202S Oscilloscope. Anyone found additional solutions?

**### lsusb output:**
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 8087:0029 Intel Corp. 
Bus 005 Device 002: ID 27c6:521d Shenzhen Goodix Technology Co.,Ltd. FingerPrint
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
**Bus 003 Device 006: ID 5345:1234 Owon PDS6062T Oscilloscope**
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**### Running owon-vds-tiny via terminal**
env: Linux, Java 11.0.21 amd64
app dir: /opt/owon-vds-tiny
user dir: /home/user1/.owon-vds-tiny
locale: en, en_US, false
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out
[Send @MACHINE_TYPE_ADD 0x56]LibusbJava.usb_bulk_read: Connection timed out

florentbr, 2023-12-31:
@wandering3ngineer
> Similar problem wtih Ubuntu Linux 20.04. GUI interface shows **OFFLINE** in upper left corner. **No available usb found** ... I'm connecting HDS2202S Oscilloscope. Anyone found additional solutions?
> 
No issue here, the HDS2202S is not a VDS1022 oscilloscope.

Petertechmail, 2024-01-03:
(RESOLVED) 
a faulty internal USB Connector, as you can see in the photo the USB connector made contact when the cable was pushed right at an angle. While the VDS enclosure was closed the end covers prevented movement of the USB connector thus giving the impression that it was a solid connection. However upon opening the VDS enclosure to make power measurements oila' power was established as the USB connector was able to move sideways. 

So I will determine if it only needs soldering or replacing. Thank you everyone for their time and patience. Sometime the issue is so simple yet escapes the obvious!
[image]