florentbr/OWON-VDS1022 #89: Yosemite 10.10.5 App opening then crashes

Petertechmail, 2023-12-05:
[image] in Yosemite. Here's a log launched from the App's Contents MacOS Terminal Launch:

MacBookPro-13:~ Peter$ /Applications/OWON\ VDS1022.app/Contents/MacOS/launch ; exit;
env: Mac OS X, Java 1.8.0_392 x86_64
app dir: /Applications/OWON VDS1022.app/Contents/Resources
user dir: /Users/Peter/Library/Application Support/OWON VDS1022
locale: en_US, en_US, true
dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
  Referenced from: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _clock_gettime
  Referenced from: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

/Applications/OWON VDS1022.app/Contents/MacOS/launch: line 3:  1510 Trace/BPT trap: 5       /usr/libexec/java_home --exec java -Xdock:name='OWON VDS1022' -Xdock:icon=icon.icns -cp 'lib/*' com.owon.vds.tiny.Main
logout

Petertechmail, 2023-12-05:
[image]

Petertechmail, 2023-12-05:
Process:               java [1631]
Path:                  /Library/Java/JavaVirtualMachines/temurin-8.jdk/Contents/Home/bin/java
Identifier:            owon-vds-tiny
Version:               1.0 (1.0)
Code Type:             X86-64 (Native)
Parent Process:        bash [1630]
Responsible:           bash [1630]
User ID:               501

Date/Time:             2023-12-05 14:03:39.589 +0100
OS Version:            Mac OS X 10.10.5 (14F2511)
Report Version:        11
Anonymous UUID:        BC136C43-AAE6-9457-C9EA-623AD30F336D

Time Awake Since Boot: 3600 seconds

Crashed Thread:        31  Java: Thread-2

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000

Dyld Error Message:
  Symbol not found: _clock_gettime
  Referenced from: /Applications/OWON VDS1022.app/Contents/Resources/lib/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

Thread 0:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   com.apple.AppKit              	0x00007fff87ae03bb -[NSWindow frame] + 1
1   com.apple.AppKit              	0x00007fff87bd1f49 -[NSEvent _initWithCGSEvent:eventRef:] + 465
2   com.apple.AppKit              	0x00007fff87b07f66 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 616
3   libosxapp.dylib               	0x000000011ffb2cde -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 121
4   com.apple.AppKit              	0x00007fff87afdaf3 -[NSApplication run] + 594
5   libosxapp.dylib               	0x000000011ffb2ae1 +[NSApplicationAWT runAWTLoopWithApp:] + 151
6   libawt_lwawt.dylib            	0x000000011ff150ac -[AWTStarter starter:] + 856
7   com.apple.Foundation          	0x00007fff87556da0 __NSThreadPerformPerform + 293
8   com.apple.CoreFoundation      	0x00007fff9160f821 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
9   com.apple.CoreFoundation      	0x00007fff916019ad __CFRunLoopDoSources0 + 269
10  com.apple.CoreFoundation      	0x00007fff91600fdf __CFRunLoopRun + 927
11  com.apple.CoreFoundation      	0x00007fff916009f8 CFRunLoopRunSpecific + 296
12  java                          	0x0000000103d63b1f CreateExecutionEnvironment + 869
13  java                          	0x0000000103d60144 JLI_Launch + 1862
14  java                          	0x0000000103d66db2 main + 76
15  libdyld.dylib                 	0x00007fff91b7f5c9 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8472348a __semwait_signal + 10
1   java                          	0x0000000103d642e0 ContinueInNewThread0 + 129
2   java                          	0x0000000103d62967 ContinueInNewThread + 170
3   java                          	0x0000000103d60dab JLI_Launch + 5037
4   java                          	0x0000000103d66db2 main + 76
5   java                          	0x0000000103d6494b apple_main + 84
6   libsystem_pthread.dylib       	0x00007fff87a2105a _pthread_body + 131
7   libsystem_pthread.dylib       	0x00007fff87a20fd7 _pthread_start + 176
8   libsystem_pthread.dylib       	0x00007fff87a1e3ed thread_start + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff84724232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff8ea1526e _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff84723136 __psynch_cvwait + 10
1   libjvm.dylib                  	0x00000001048a0fb9 os::PlatformEvent::park() + 181
2   libjvm.dylib                  	0x000000010489016b ObjectMonitor::wait(long, bool, Thread*) + 753
3   libjvm.dylib                  	0x0000000104947c75 ObjectSynchronizer::wait(Handle, long, Thread*) + 251
4   libjvm.dylib                  	0x00000001047448cf JVM_MonitorWait + 327
5   ???                           	0x000000010527e087 0 + 4381466759
6   ???                           	0x000000010526fffd 0 + 4381409277
7   ???                           	0x000000010526fffd 0 + 4381409277
8   ???                           	0x000000010526fffd 0 + 4381409277
9   ???                           	0x0000000105270042 0 + 4381409346
[... 869 lines cut ...]
__TEXT                            144.7M
__UNICODE                           552K
mapped file                         1.4G
shared memory                        68K
===========                      =======
TOTAL                               7.3G
TOTAL, minus reserved VM space      7.3G
 

Model: MacBookPro7,1, BootROM 68.0.0, 2 processors, Intel Core 2 Duo, 2.4 GHz, 16 GB, SMC 1.62f7
Graphics: NVIDIA GeForce 320M, NVIDIA GeForce 320M, PCI, 256 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1067 MHz, 0x85F7, 0x483634314755363746373036364700000000
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1067 MHz, 0x85F7, 0x483634314755363746373036364700000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 4.3.6f4 17916, 3 services, 19 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Hitachi HTS545025B9SA02, 250.06 GB
Serial ATA Device: MATSHITADVD-R   UJ-898
USB Device: Internal Memory Card Reader
USB Device: Built-in iSight
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: IR Receiver
USB Device: Apple Internal Keyboard / Trackpad
Thunderbolt Bus:

florentbr, 2023-12-05:
The issue is with the provided `libusb-1.0.0.dylib` which is likely not compatible with earlier versions of MacOS. Try an install of  libusb from brew if yosemite is still supported. I would also try the `libusb-1.0.0.dylib` from a previous release: https://github.com/florentbr/OWON-VDS1022/tree/1.0.30-cf10/lib/mac/x86_64

Petertechmail, 2023-12-05:
I assume within the Contents of the OWON App in Applications. I simply replace this file with an earlier version which you suggest. Correct me if I’m wrong.

Petertechmail, 2024-01-03:
this is the contents of the App:
[image]

Petertechmail, 2024-01-03:
Thanks! that did it! replaced both from link above:
libusb-1.0.0.dylib
libusb-0.1.4.dylib

(RESOLVED) older libraries replaced