florentbr/OWON-VDS1022 #94: Bad flash header: 0xFFFF

yurii-vavrynchuk-embrox, 2024-02-16:
Hello, I have some problem with flash memory.
[image]

This problem was happened after I tried to load firmware from the C++ (according to your algorithms).
I have no ability to perform factory reset (nothing happens after that).
Also I can't run your python code because of the problem above.

Please, help!

florentbr, 2024-02-16:
Looks like the flash is corrupted.
It's not that serious since the flash only holds the calibration and version, not the firmware.
It's quite simple to write it back with the python API.

```python
device = VDS1022(flash="backup-flash.bin")
device.sync_flash()
```
If you previously attempted to overwrite the flash with the python API, then there's a backup in the current folder.
If you don't have a backup, then build the `flash.bin` from spec (see issue #14).

The official software automatically saves the flash parameters in a file when launched.
It should be in a `flash_txt` folder.

Regarding the algorithm, you may have to change the header from `55 AA` to `AA 55` with newer devices.
https://github.com/florentbr/OWON-VDS1022/blob/4c67805713906c20b4414b4225fd293adea4cb05/api/python/vds1022/vds1022.py#L1689

yurii-vavrynchuk-embrox, 2024-02-19:
Hello, @florentbr.
Thank you so much for your answer so much!
Best regards,
Yurii